me-no-dev/ESPAsyncWebServer #108: Trouble with request headers and Internet Explorer

bsbweliton, 2016-12-09:
Guys, i'm working on application that uses custom header. The "request->headers()" is not finding it and others headers in IE. On Chrome finds almost entire headers, including mine.
For purpose of test I made a function that return headers obtained by apllication:
  
`
    int headers = request->headers();
    int i;
    String retorno = "";
    for(i=0;i<headers;i++)
    {
          AsyncWebHeader* h = request->getHeader(i);
          retorno += h->name().c_str();
          retorno += ": ";
          retorno += h->value().c_str();
          retorno += "#";
     } 
     AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", retorno.c_str());
        response->addHeader("Server","ESP Async Web Server");
        request->send(response);  
`

in Chrome shows:
[image]
Connection: keep-alive
Accept: text/html
Origin: http://192.168.25.109
X-Requested-With: XMLHttpRequest
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/54.0.2840.99 Safari/537.36
auth: dGVzdDp0ZXN0
Referer: http://192.168.25.109/
Accept-Encoding: gzip, deflate
Accept-Language: pt-BR,pt;q=0.8,en-US;q=0.6,en;q=0.4
Cookie: _ga=GA1.1.583080974.1481303502; _gat=1
1

in IE shows:
[image]
Connection: Keep-Alive
Cache-Control: no-cache
Cookie: _ga=GA1.1.5916540.1481304000
 _gat=1
1304000; _gat=1

bsbweliton, 2016-12-09:
[Update]On Firefox and Opera works like a charm. The problem is only on IE.

bsbweliton, 2016-12-10:
I discovered the line on WebRequest.cpp. I made a workaround: I removed the condition on line 297. `if(_interestingHeaders->containsIgnoreCase(name) || _interestingHeaders->containsIgnoreCase("ANY")) `
Works fine.

me-no-dev, 2016-12-13:
hmm... it should have worked. That condition you have removed is there to filter the headers. How do you assign your handler?

bsbweliton, 2016-12-13:
> hmm... it should have worked. That condition you have removed is there to filter the headers. How do you assign your handler?

Follow my handler, I use the "X-auth" header to a custom authentication:
`  
g_server.on("/index.html", [](AsyncWebServerRequest *request){ 

    if(request->hasHeader("X-Auth")){
      if(checAuthentication(request->header("X-Auth").c_str(),"teste","teste"))
        request->send(SPIFFS, "/normal.html","text/html");
      else
      {
        AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Erro");
        response->addHeader("Server","ESP Async Web Server");
        request->send(response);  
      }
      
    }else
    {        
      AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Erro");
      response->addHeader("Server","ESP Async Web Server");
      request->send(response);             
    }
  
  });`

bsbweliton, 2016-12-13:
One doubt: what is the purpose of filter the headers?

me-no-dev, 2016-12-13:
save memory :) give me a few and I'll write something for you

me-no-dev, 2016-12-13:
```cpp
g_server.on("/index.html", [](AsyncWebServerRequest *request){
  if(request->hasHeader("X-Auth")){
    if(checAuthentication(request->header("X-Auth").c_str(),"teste","teste"))
      request->send(SPIFFS, "/normal.html","text/html");
    else
    {
      AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Erro");
      response->addHeader("Server","ESP Async Web Server");
      request->send(response);  
    }
  }else
  {        
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Erro");
    response->addHeader("Server","ESP Async Web Server");
    request->send(response);             
  }
}).setFilter((AsyncWebServerRequest *request) {
  request->addInterestingHeader("X-Auth");
  return true;
});
```

me-no-dev, 2016-12-13:
this is a problem with IE and Microsoft. Standards dictate that the "Host" header must be the first header and IE does not do that, the result is that the command for collecting all headers is executed after the headers have started to come and you are getting only the ones that are after "Host"

me-no-dev, 2016-12-13:
hmm... the above filter will not work either

bsbweliton, 2016-12-13:
Thanks so much, but the `addInterestingHeader` method doesn't exists yet. It will be implemented?

me-no-dev, 2016-12-13:
it does exist :) https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L205

bsbweliton, 2016-12-13:
You're right. Sorry. I got confused.

universam1, 2017-04-19:
Could this be related #94 to websockets not working on Edge and IE?

forkineye, 2017-06-23:
I ran into this last year and my simple hack to handle it is here - https://github.com/forkineye/ESPAsyncWebServer/commit/ace81f7c64150f1f19e6afa38b9b28e236e24433.  @me-no-dev stated at the time it may cause other issues in the code, but I can't recall what those were.  It's been working fine for me.

universam1, 2017-06-23:
@me-no-dev would it be possible to include this https://github.com/forkineye/ESPAsyncWebServer/commit/ace81f7c64150f1f19e6afa38b9b28e236e24433 patch for to solve this issue finally?

me-no-dev, 2017-06-23:
@universam1 issue with that is that you can not branch based on host. The way to make it work is to collect all headers and deal with them once done. Downside is obviously more memory usage with every request and some browsers send way more headers than necessary.

universam1, 2017-06-23:
thanks @me-no-dev for the explanation! 
I would see a solution in implement an optional IE compatible mode in case the someone (like me) is dependent on, could be enabled by a define.

closed 2019-10-05 by stale[bot]