me-no-dev/ESPAsyncWebServer #109: Iterate over the list of WebSocket clients

rakeshpai, 2016-12-10:
Is it possible to get a list of the currently connected websocket clients?

Say for eg. that I'm trying to implement this: Every minute, send temperature data to all connected STA clients, but not to any of the AP clients.

Given a client object, I can tell if it's connected over STA or AP by checking IP addresses, but I can't find a way to enumerate/iterate over all the connected clients. Is there a way to do this?

I see that you are maintaining a [list of clients](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.h#L142), but this is a private member. Is there any way of accessing or iterating over this list externally? If not, could you consider making the list public in some way?

Thanks for the great lib!

DedeHai, 2016-12-11:
+1 for this!
the pubsub libary has a command to 'broadcast' to all connected clients, maybe add something like that with an added filter for STA or AP?

me-no-dev, 2016-12-13:
maybe some sort of getter and iterator like the ones used for listing files and WiFi access points?
adding more methods to all clients is not something I really want to do. There are more then enough methods now :(

rakeshpai, 2016-12-14:
Works for me. :+1:

palsbo, 2018-12-31:
I am missing this option too. In the good old SocketIO, a broadcast would send to all clients but the one issuing the broadcast. That was nice. 
Example:
A slider on a page sends update to be broadcasted to other clients.
When coming back as a broadcast the slider may already have changed again and the whole thing starts going wild.
Of course - there are ways to avoid this on the client, but this is clumsy!
I made a addition to the textAll function by supplying a client and prevent the loop from sending to that client.
It works, but I would prefer not keep changing for each new update.

closed 2019-10-05 by stale[bot]