me-no-dev/ESPAsyncWebServer #116: background interrupt task?

mark-hahn, 2017-01-09:
I want to run some code in a separate *task*, running off interrupts.  An example would be blinking an LED while waiting for a station connection.  I know that there are only 3 interrupt OS tasks possible in the SDK.  I assume the async webserver and or other code is using them all.

How can I do this?

me-no-dev, 2017-01-10:
Async networking is using the network interrupt so you are safe to use all others :) timers, tickers whatever you need :)

mark-hahn, 2017-01-10:
Thanks.  I thought there were only three total and I assumed the wifi used one.  I guess the async web server uses the same int as wifi.  I don't know what a timer or ticker is so I'll have to research this.

closed 2017-01-10 by mark-hahn