me-no-dev/ESPAsyncWebServer #118: ESP32 Support

williamesp2015, 2017-01-14:
Hi, I would like to know how can use ESPAsyncWebserver with ESP32 core.
Thank you

Matheus-Garbelini, 2017-01-15:
Man, it would be awesome if me-no-dev ported this to esp32, but unfortunattely, there are a lot of code that me-no-dev depend to make it happen, Luke SPIFFS for instance. God, the current esp32 is só small, se just have 3 or 4 libraries. The main hardware features like complete spi, i2c are missing and just happen to be in the original SDK. So for now I recommend you to use VisualGDB extension + visual studio to lightning faster develop to ESP32. Its not all that hard using SDK function, they are well documented

me-no-dev, 2017-01-16:
I am somewhat working on it. Issue is mainly the different approach required. Will not be even as true async as it is on ESP8266, because we do not have access to that interrupt and connections will be polled in a thread in order to get it working. Still first come- first serve, so multiconnection will work, just not quire the same way.
New FS classes are in need :)

hreintke, 2017-01-16:
Does that mean there is no RAW LWIP available in ESP32 or are there other limitations ?

me-no-dev, 2017-01-16:
there is raw lwip but you can not run your code in it's context. You have to take the event and then dispatch it in a thread. [this](https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L179-L258) is how I did it for the just-merged new mDNS service. While UDP is quite easy to pull, TCP is another beast :) it has many and different events and I need to spend some thinking over how to properly queue them,

Matheus-Garbelini, 2017-01-17:
@me-no-dev fantastic. I remember the time when you conceive-us the spi slave library when no one was able to do it. It seems that arduino community are very difficult to work with native registers and pass all day reading the core documentation like you do. Congratulations body. Your project is today more stable and contain more features than Sming does. Keep going the good work.

I,m not getting my esp32 yet because they are yet expensive, but when they drop the price just a little I will be able do navigate trough its madness haha

me-no-dev, 2017-01-17:
@Matheus-Garbelini thanks for the kind words :) ESP32 is still far behind on many things :) We have so much new stuff in it that it will take a while (not that ESP8266 didn't).

hreintke, 2017-01-17:
@me-no-dev : 
You are almost writing a local socket implementation in this way.
Not knowing the full ESP32 environment so do know the exact definition/usage of a context but what would be the limitation for the code to be executed in a (lwip) callback.

me-no-dev, 2017-01-17:
in ESP8266 we get the callbacks called through interrupt, so regardless of what our sketch is doing, packets are handled as soon as they come.
ESP32 is like a computer with 2 cores and operating system and threads. The callbacks do come but they are called from another thread, not the interrupt and we are handling them in a separate thread than that (or loop thread for regular clients or separate thread for async), so we will get the packet after it has been given to the lower thread and a couple of context switches happen in RTOS so that threads are run

williamesp2015, 2017-02-28:
Is there any hope to get ESPAsyncTCP compatible with ESP32? I got Errors:

src\ESPAsyncTCP.cpp:182:96: error: 'ip_addr_t {aka struct _ip_addr}' has no member named 'addr'
return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) && (_pcb->remote_port == other._pcb->remote_port));
^
src\ESPAsyncTCP.cpp:457:6: error: prototype for 'void AsyncClient::_dns_found(ip_addr_t*)' does not match any in class 'AsyncClient'
void AsyncClient::_dns_found(ip_addr_t *ipaddr){

me-no-dev, 2017-02-28:
Will work in the future in ESP32, but can not work in the same way as it does on ESP8266. You can not run the entire server code in the LwIP thread on ESP32 because it does not have enough memory given to it (and it should be that way), So i need to spin a thread to deal with network events and propagate them to the clients and servers attached. I am thinking about it often in order to prepare all steps needed in my head for this to actually work reliably. Hod your horses and you will get AsyncESP on ESP32 :)

williamesp2015, 2017-03-04:
My horse is ready to test AsyncESP. Is there any progress to help many?

me-no-dev, 2017-03-05:
no, but I have spent more time thinking about it and testing other network stuff. Maybe soon I'll have a better picture.

williamesp2015, 2017-03-27:
@me-no-dev 
Any progress on ESP32?

Matheus-Garbelini, 2017-04-03:
@williamesp2015 Calm down dude, haha. Esp32 Arduino core just received few FS support, and currently, @me-no-dev updated Support to self-signed certificates. There's a lot of background processing before an async library is done.

ZIMkaRU, 2017-08-22:
Any progress on ESP32?

me-no-dev, 2017-09-08:
Done! Requires still SPIFFS to be merged into IDF and libs rebuild but that should happen in the next few days

closed 2017-09-08 by me-no-dev

nardev, 2017-09-08:
@me-no-dev can you push it somewhere?

me-no-dev, 2017-09-09:
@nardev works exactly the same way ;) just different AsyncTCP lib for ESP32

rmhomecouk, 2017-09-16:
hey @me-no-dev - I've tried the code and a couple of things I've noticed from using the example code ("ESP_AsyncFSBrowser"). Not I'm using the #include <AsyncTCP.h> file for this too.

1: The "events" don't work - I get 404 rather than the stream
2: The unit reboots due to an "abort()" down to corrupt heap.

I've tried on multiple units, same issue.

---------------- SERIAL OUTPUT

CORRUPT HEAP: Bad tail at 0x3ffdfe5a. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c", line 195, function: multi_heap_free
abort() was called at PC 0x400ddceb on core 1

Backtrace: 0x400872e0:0x3ffdf300 0x400873df:0x3ffdf320 0x400ddceb:0x3ffdf340 0x40087009:0x3ffdf370 0x40082602:0x3ffdf390 0x40082a6d:0x3ffdf3b0 0x4000bec7:0x3ffdf3d0 0x400d9d27:0x3ffdf3f0 0x400d8a75:0x3ffdf440 0x400d7ea2:0x3ffdf480 0x400d8987:0x3ffdf4c0 0x400d5bdd:0x3ffdf510 0x400d85a7:0x3ffdf5d0 0x400d8611:0x3ffdf610 0x400d87f5:0x3ffdf650 0x400d4999:0x3ffdf670 0x400d4bc5:0x3ffdf6c0

Rebooting...
ets Jun  8 2016 00:22:57

---------------- DECODED STACK TRACE

Decoding 17 results
0x400872e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400873df: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400ddceb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x40087009: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c line 270
0x40082602: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 172
0x40082a6d: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 33
0x400d9d27: AsyncAbstractResponse::_ack(AsyncWebServerRequest*, unsigned int, unsigned int) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebResponses.cpp line 681
0x400d8a75: AsyncAbstractResponse::_respond(AsyncWebServerRequest*) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebResponses.cpp line 681
0x400d7ea2: AsyncWebServerRequest::send(AsyncWebServerResponse*) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 975
0x400d8987: AsyncWebServerRequest::send(fs::File, String const&, String const&, bool, std::function ) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 975
0x400d5bdd: SPIFFSEditor::handleRequest(AsyncWebServerRequest*) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/SPIFFSEditor.cpp line 420
0x400d85a7: AsyncWebServerRequest::_parseLine() at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 975
0x400d8611: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 975
0x400d87f5: std::_Function_handler ::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at C:\Dropbox\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 975
:  (inlined by) _M_invoke at c:\program files (x86)\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0/functional line 1871
0x400d4999: std::function ::operator()(void*, AsyncClient*, void*, unsigned int) const at C:\Dropbox\Arduino\libraries\AsyncTCP\src/AsyncTCP.cpp line 743
:  (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at C:\Dropbox\Arduino\libraries\AsyncTCP\src/AsyncTCP.cpp line 497
0x400d4bc5: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at C:\Dropbox\Arduino\libraries\AsyncTCP\src/AsyncTCP.cpp line 743
:  (inlined by) _handle_async_event at C:\Dropbox\Arduino\libraries\AsyncTCP\src/AsyncTCP.cpp line 80
:  (inlined by) _async_service_task at C:\Dropbox\Arduino\libraries\AsyncTCP\src/AsyncTCP.cpp line 96

me-no-dev, 2017-09-18:
@rmhomecouk interesting... events work here :) maybe something else is wrong. Also that Heap error could be due to some sdkconfig value. I would need to investigate further. Here is my Async test code as I am running it on ESP32:
```cpp

#include "ESPAsyncWebServer.h"
#include "SPIFFSEditor.h"
AsyncWebServer server(80);
AsyncWebSocket ws("/ws");
AsyncEventSource events("/events");

void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
  if(type == WS_EVT_CONNECT){
    Serial.printf("ws[%s][%u] connect\n", server->url(), client->id());
    client->printf("Hello Client %u :)", client->id());
    client->ping();
  } else if(type == WS_EVT_DISCONNECT){
    Serial.printf("ws[%s][%u] disconnect\n", server->url(), client->id());
  } else if(type == WS_EVT_ERROR){
    Serial.printf("ws[%s][%u] error(%u): %s\n", server->url(), client->id(), *((uint16_t*)arg), (char*)data);
  } else if(type == WS_EVT_PONG){
    Serial.printf("ws[%s][%u] pong[%u]: %s\n", server->url(), client->id(), len, (len)?(char*)data:"");
  } else if(type == WS_EVT_DATA){
    AwsFrameInfo * info = (AwsFrameInfo*)arg;
    String msg = "";
    if(info->final && info->index == 0 && info->len == len){
      //the whole message is in a single frame and we got all of it's data
      Serial.printf("ws[%s][%u] %s-message[%llu]: ", server->url(), client->id(), (info->opcode == WS_TEXT)?"text":"binary", info->len);

      if(info->opcode == WS_TEXT){
        for(size_t i=0; i < info->len; i++) {
          msg += (char) data[i];
        }
      } else {
        char buff[3];
        for(size_t i=0; i < info->len; i++) {
          sprintf(buff, "%02x ", (uint8_t) data[i]);
          msg += buff ;
        }
      }
      Serial.printf("%s\n",msg.c_str());

      if(info->opcode == WS_TEXT)
        client->text("I got your text message");
      else
        client->binary("I got your binary message");
    } else {
      //message is comprised of multiple frames or the frame is split into multiple packets
      if(info->index == 0){
        if(info->num == 0)
          Serial.printf("ws[%s][%u] %s-message start\n", server->url(), client->id(), (info->message_opcode == WS_TEXT)?"text":"binary");
        Serial.printf("ws[%s][%u] frame[%u] start[%llu]\n", server->url(), client->id(), info->num, info->len);
      }

      Serial.printf("ws[%s][%u] frame[%u] %s[%llu - %llu]: ", server->url(), client->id(), info->num, (info->message_opcode == WS_TEXT)?"text":"binary", info->index, info->index + len);

      if(info->opcode == WS_TEXT){
        for(size_t i=0; i < info->len; i++) {
          msg += (char) data[i];
        }
      } else {
        char buff[3];
        for(size_t i=0; i < info->len; i++) {
          sprintf(buff, "%02x ", (uint8_t) data[i]);
          msg += buff ;
        }
      }
      Serial.printf("%s\n",msg.c_str());

      if((info->index + len) == info->len){
        Serial.printf("ws[%s][%u] frame[%u] end[%llu]\n", server->url(), client->id(), info->num, info->len);
        if(info->final){
[... 81 lines cut ...]
          Serial.printf("_POST[%s]: %s\n", p->name().c_str(), p->value().c_str());
        } else {
          Serial.printf("_GET[%s]: %s\n", p->name().c_str(), p->value().c_str());
        }
      }

      request->send(404);
    });
    server.onFileUpload([](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final){
      if(!index)
        Serial.printf("UploadStart: %s (%lu)\n", filename.c_str(), millis());
      //Serial.printf("%s", (const char*)data);
      if(final)
        Serial.printf("UploadEnd: %s (%u) (%lu)\n", filename.c_str(), index+len, millis());
    });
    server.onRequestBody([](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
      if(!index)
        Serial.printf("BodyStart: %u\n", total);
      Serial.printf("%s", (const char*)data);
      if(index + len == total)
        Serial.printf("BodyEnd: %u\n", total);
    });
    server.begin();
}
```

rmhomecouk, 2017-09-18:
Hey @me-no-dev  thanks for the reply - OK, I tried a copy/paste of your code (with the WiFi begin etc) and the same thing - events returns 404 and same HEAP crash when using the FSEditor. I will try a make erase and see if I can clear the whole unit - though this is unit 4 and all appear the same. FYI I'm also using the latest commits of the Arduino code and this library. I did this in order to get the latest additions for SPIFFS support.

cheers
[001.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/1310215/001.txt)

rmhomecouk, 2017-09-18:
ok I got the latest commits and did a "flash_erase" - same issue I'm afraid :(

CORRUPT HEAP: Bad tail at 0x3ffdd01a. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c", line 195, function: multi_heap_free
abort() was called at PC 0x400dc8eb on core 1

Backtrace: 0x40087278:0x3ffdb690 0x40087377:0x3ffdb6b0 0x400dc8eb:0x3ffdb6d0 0x40086fa1:0x3ffdb700 0x400825ca:0x3ffdb720 0x40082a35:0x3ffdb740 0x4000bec7:0x3ffdb760 0x400d840f:0x3ffdb780 0x400d6f35:0x3ffdb7d0 0x400d6272:0x3ffdb810 0x400d6e47:0x3ffdb850 0x400d3fad:0x3ffdb8a0 0x400d6a67:0x3ffdb960 0x400d6ad1:0x3ffdb9a0 0x400d6cb5:0x3ffdb9e0 0x400da0b5:0x3ffdba00 0x400da2e1:0x3ffdba50

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:732
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 7 - STA_GOT_IP
.192.168.92.218
44.44°C
NOT_FOUND: GET http://192.168.92.218/events
_HEADER[Host]: 192.168.92.218
_HEADER[Connection]: keep-alive
_HEADER[Authorization]: Digest username="admin", realm="asyncesp", nonce="795d0e2e77f0bd1ac56d88a223e30cc8", uri="/events", response="d807462f47ede90468923546c43a6743", opaque="0bc073c1bf61fa0ae678fa5892cfd2a6", qop=auth, nc=00000049, cnonce="33cd4e8ca0b532d9"
_HEADER[User-Agent]: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.91 Safari/537.36
_HEADER[Upgrade-Insecure-Requests]: 1
_HEADER[Accept]: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8
_HEADER[DNT]: 1
_HEADER[Accept-Encoding]: gzip, deflate
_HEADER[Accept-Language]: en
[W][AsyncTCP.cpp:528] _poll(): rx timeout 4
[W][AsyncTCP.cpp:528] _poll(): rx timeout 4

PS, thanks for all your hard work with this! truly amazing work! 👍

me-no-dev, 2017-09-18:
OK I got that same exception if the index file does not exist. As soon as I added index.htm it stopped giving the exception, so it is probably related to the file system or something (will investigate further) but using the index from the example made everything work. That includes events :)
Rather strange why that exception on missing file though...

me-no-dev, 2017-09-18:
[image]

etiquet, 2017-09-18:
Top ! ( super, great, marvelous, bravo ! )

ZIMkaRU, 2017-09-19:
Thank you very much!)

rmhomecouk, 2017-09-19:
Hey @me-no-dev thanks for the update and glad that it's reproducible and not just me :) As you say, adding a file manually (check at boot-up if it exists or not) works around it for now! 

cheers!

me-no-dev, 2017-09-19:
we have found an issue in SPIFFS itself that might fix those issues :) will try to update all tools and boards tomorrow :) Do not use files above 25KB on spiffs yet ;)

me-no-dev, 2017-09-22:
alright SPIFFS has been updated :) give things a go now

rmhomecouk, 2017-11-14:
hey @me-no-dev I just wanted to say thanks very much for sorting this! I've finally managed to update my LoRa gateways and I can confirm events and file uploads is now working properly.

thanks again!

me-no-dev, 2017-11-16:
you are welcome!

Albotech, 2017-11-17:
I stumbled on this issue when I added the ESPAsyncWebServer library in my ESP32 project.

Here two cases when it cause the ESP32 to crash and reset:
1) An empty file (0 byte) (Example:  FILE: /css/app.css  SIZE: 0)
2) A missing file where an html refers to it. (Example: <link rel="stylesheet" href="css/app.css"> but the file doesn't exists in SPIFF filesystem)

So, for the moment, be sure the files exists and are not empty in your SPIFF filesystem.

Thanks for your good work!

treii28, 2018-05-01:
Wow, yeah - just tried to compile this in platform io just to see if it would - I had to copy about 2 dozen files over from the esp8266 core just to get around the include errors and then got about 12 pages of compiler errors. What's the status of porting this over?