me-no-dev/ESPAsyncWebServer #123: How can i access the body of POST request and can decode the JSON string?

ovi1337, 2017-02-03:
The ESP8266WebServer is supporting `String result = server->arg("plain");` which is have the plain body of the POST request. I don't have found any example or it's not clear for me, now i can access the content of the message-body inside of an event.

```c
server.on("/api/test", HTTP_GET, [](AsyncWebServerRequest *request) {
    String result = request->arg("plain")->value().c_str();
    Serial.println(result);

    StaticJsonBuffer<50> jsonBuffer;
    JsonObject& data = jsonBuffer.parseObject(server->arg("plain"));  

    boolean enabled = data["enabled"];
    Serial.println(enabled ? "true" : "false");

    if(request->hasParam("foo")) {
      AsyncWebParameter* p = request->getParam("foo");
      Serial.printf(p->value().c_str());
    }

    request->send(200, "text/plain", "");
  });
```

Another thing which is relating to this question is, how can i access the body to be able to decode the sent JSON?
Can you please give me a small help for it?

Could it be that the whole part is missing in your documentation? I only have seen, how i can send a JSON object, but nothing how i can receive and decode it.

Thank you in advance!

me-no-dev, 2017-02-03:
the request plain body comes in pieces in Async, while it's collected in the regular server.
Those pieces are collected through the ```onRequestBody``` callbacks.
https://github.com/me-no-dev/ESPAsyncWebServer#body-data-handling
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L381
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L387

ovi1337, 2017-02-03:
awesome buddy, now i've got it, thx! ^^
i didn't have seen, that it's a second/third callback...

closed 2017-02-03 by ovi1337

sochs, 2017-04-06:
I have just the same probem. Could you briefly describe your solution? That would be great!

probonopd, 2017-07-25:
Can you please post a working example?

sochs, 2017-07-25:
Try this:

`server.onRequestBody([](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
      if (request->url() == "/test") {
        if (!handleTest(request, data)) request->send(200, "text/plain", "false");
          request->send(200, "text/plain", "true");
}
});`

`bool handleTest(AsyncWebServerRequest *request, uint8_t *datas) {

  Serial.printf("[REQUEST]\t%s\r\n", (const char*)datas);
  
  DynamicJsonBuffer jsonBuffer;
  JsonObject& _test = jsonBuffer.parseObject((const char*)datas); 
  if (!_test.success()) return 0;

  if (!_test.containsKey("command")) return 0;
  String _command = _test["command"].asString();
  Serial.println(_command);
  
  return 1;
}`

probonopd, 2017-07-25:
Thanks for offering your help, can't get it to work either.

```

#include <ESP8266WiFi.h>

#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h> // https://github.com/alanswx/ESPAsyncWiFiManager/

#include "AsyncJson.h"
#include "ArduinoJson.h"

AsyncWebServer server(80);
DNSServer dns;

bool handleTest(AsyncWebServerRequest *request, uint8_t *datas) {

  Serial.printf("[REQUEST]\t%s\r\n", (const char*)datas);
  
  DynamicJsonBuffer jsonBuffer;
  JsonObject& _test = jsonBuffer.parseObject((const char*)datas); 
  if (!_test.success()) return 0;

  if (!_test.containsKey("command")) return 0;
  String _command = _test["command"].asString();
  Serial.println(_command);
  
  return 1;
}

void setup() {

  Serial.begin(115200);

  AsyncWiFiManager wifiManager(&server, &dns);
  wifiManager.autoConnect("AutoConnectAP");

  server.onRequestBody([](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
    Serial.println("Running");
    if (request->url() == "/test") {
      if (!handleTest(request, data)) request->send(200, "text/plain", "false");
      request->send(200, "text/plain", "true");
    }
  });
  server.begin();
}

void loop() {
}
```

When I call it with
```
curl -H "Content-Type: application/json" -X POST -d '{"command": "Hello"}' http://192.168.0.22/
```
it prints "Running" on serial but that's it.

probonopd, 2017-07-25:
Thank you so much, my friend. This seems to be working!

Danial-Kord, 2021-11-14:
My issue was with my js client which is weird look at this:
After HOURS finally I can send data to ESP
what was challenge?
good to ask:
this wont work!
(
var params = 'value=';
var val = document.getElementById("value").innerHTML;
params = params.concat(val);
params = params.concat('&name=binny');

http.send(params);
)

how ever this works fine :/
(
var val = document.getElementById("value").innerHTML;
var params = `orem=${val}&name=binny`;

http.send(params);
)
first one doesn't send 'params' in right format! but second one does
I am confused but finally relived

RezaAzhdarkosh, 2023-03-22:
It is very easy to receive information in json form when it is sent as post or put.
I will write a simple example for you to receive the temperature and humidity information that I send as json codes using the put method:

server.onRequestBody([](AsyncWebServerRequest * request, uint8_t *data, size_t len, size_t index, size_t total)
  {
    if (request->url() == "/api/update")
    {
      StaticJsonDocument<256> doc;
      deserializeJson(doc, data);
      uint8_t SLAVE_ID = doc["id"].as<uint8_t>();
      uint8_t TEMP = doc["temp"].as<uint8_t>();
      uint8_t HUM = doc["hum"].as<uint8_t>();
      Serial.println(SLAVE_ID);
      Serial.println(TEMP );
      Serial.println(HUM );
  } });

Tombstone2K, 2023-07-19:
> Try this:
> 
> ```
> server.onRequestBody([](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
>   if (request->url() == "/test") {
>     if (!handleTest(request, data)) request->send(200, "text/plain", "false");
>     request->send(200, "text/plain", "true");
>   }
> });
> ```
> 
> and
> 
> ```
> bool handleTest(AsyncWebServerRequest *request, uint8_t *datas) {
> 
>   Serial.printf("[REQUEST]\t%s\r\n", (const char*)datas);
>   
>   DynamicJsonBuffer jsonBuffer;
>   JsonObject& _test = jsonBuffer.parseObject((const char*)datas); 
>   if (!_test.success()) return 0;
> 
>   if (!_test.containsKey("command")) return 0;
>   String _command = _test["command"].asString();
>   Serial.println(_command);
>   
>   return 1;
> }
> ```

This really works. Thanks

Tombstone2K, 2023-07-19:
> It is very easy to receive information in json form when it is sent as post or put. I will write a simple example for you to receive the temperature and humidity information that I send as json [...]

This approach works too. Thanks