me-no-dev/ESPAsyncWebServer #124: Problem with Async Response Callback download

Matheus-Garbelini, 2017-02-07:
Hello, I was trying to download a file by using async response call back (I cant use the already provided option by using a stream because a i need to do something else. The code is the following:

Callback

`AsyncWebServerResponse *response = request->beginResponse("application/octet-stream", 12, [](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
        snprintf((char *)buffer, maxLen, "test");
        return 12;
    });`

Response headers:

   `response->addHeader("Cache-Control", "no-cache");
    response->addHeader("Content-Disposition", "attachment; filename=" + String("download.txt"));
    response->addHeader("Access-Control-Allow-Origin", "*");`

I can download the file but the answer i receive is something like 

> 7465 7374

if i convert it to ascii using online converter tools, it appears as "test" as i defined in the programming. How can i fix this?

Matheus-Garbelini, 2017-02-10:
@me-no-dev can you reproduce this issue? if not I can prepare some sample code to generate it

me-no-dev, 2017-02-10:
Use chunked response and do not return 12, why return 12 when "test" is 4 bytes long?
```cpp
AsyncWebServerResponse *response = request->beginChunkedResponse("application/octet-stream", 
    [](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
        if(index){ //already sent
            return 0;
        }
        return snprintf((char *)buffer, maxLen, "test");
    });
response->addHeader("Cache-Control", "no-cache");
response->addHeader("Content-Disposition", "attachment; filename=" + String("download.txt"));
response->addHeader("Access-Control-Allow-Origin", "*");
```

Thy this and see how it goes

Matheus-Garbelini, 2017-02-10:
Oh my god, It worked. Thank you a lot, I was almost one week trying to fix SD download.
By the way, why it happened with callback? I tought it was a good idea to use callback as i can inform the total file size.

me-no-dev, 2017-02-10:
that 12 was the problem, chunked response is just better and does not make you know the length. The browser still gets what it expects

Matheus-Garbelini, 2017-02-10:
Hum, OK, but when I actually changed 12 to for instance 4 or 3, I still got HEX values when using callback

me-no-dev, 2017-02-10:
maybe share the full response that the other end get's? else it's all guess game. That should be the only difference between chunked and not chunked response.

Matheus-Garbelini, 2017-02-10:
Ok, do you mean, the entire response including header that are sent betwen the ESP and the PC?
I will use a tool to verify that.

Matheus-Garbelini, 2017-02-10:
@me-no-dev 
        Hello, Thanks for your help, Indeed I was sending the content with a kind of lenght mismath difference. I couldn't get all the response because my browser doesn't recognize the kind of response when there are a mistake like that. Unfortunatelly, the situation in someway didn't gave me error while downloading a incomplete package and I tought it could be something with the library. I fixed the SD card callback function and the content seems to be working. I will post here how a callback to SD using async with somekind of MUTEX over serial (to share SD Card bus).

Later when everthing is stable I could provide a complete example of a SPIFFS + SDCard filesystem with this library if you'd like to include as an example or a related project using this library.
Thank you a lot, SD card download and upload is extremelly fast and non blocking with your library.
PS: My callback have functions that block the request even tought it's not recomended. ESP8266 doesn't have nested interrupt isn't it?

My callback:

```c++
bool loadFromSdCard(AsyncWebServerRequest *request)
{
	if (isSendingSD == 0 && ESP_Status_Data->SdcardOK == 1) {
		//DEBUG_EVENT("LOAD SD Card requested");
		if (request->url() != "/editSD") returnFail(request, "Not Found");
		if (sending == 1) returnFail(request, "Already sending");
		char name[64];
		//substitute percent20 to space
		String path = (String)request->arg("path");
		//DEBUG_EVENT("path: " + path);
		std::string path_ = path.c_str();
		ReplaceStringInPlace(path_, "%20", " "); //troca %20 por " " - espaço (correção de padding)
		path = path_.c_str();

		waitSD();
		SDReadFile = SD.open(path.c_str(), O_READ);
		leaveSD();

		if (!SDReadFile) {
			DEBUG_MESSAGE("File dont exist");
			//DEBUG_EVENT("File dont exist");
			returnFail(request, "File dont exist");
		}
		isSendingSD = 1;
		waitSD();
		SDReadFile.getName(name, 63);
		leaveSD();

		dataAvaliable = SDReadFile.size() - SDReadFile.position();
		DEBUG_MESSAGE("Total file size:" + String(dataAvaliable));

		AsyncWebServerResponse *response = request->beginResponse("text/plain", dataAvaliable,
			[](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
			uint32_t readBytes;
			uint32_t bytes = 0;
			uint32_t avaliableBytes = SDReadFile.available();

			waitSD();

			if (avaliableBytes > maxLen) {
				bytes = SDReadFile.readBytes(buffer, maxLen);
			}
			else {
				bytes = SDReadFile.readBytes(buffer, avaliableBytes);
				isSendingSD = 0;
				SDReadFile.close();
			}

			leaveSD();
			return bytes;
		});

		response->addHeader("Cache-Control", "no-cache");
		response->addHeader("Content-Disposition", "attachment; filename=" + String(name));
		response->addHeader("Access-Control-Allow-Origin", "*");
		request->send(response);
	}
	else
	{
		returnFail(request, "SD card busy or not initialized");
	}
} 
```

closed 2017-02-10 by Matheus-Garbelini