me-no-dev/ESPAsyncWebServer #125: Example/Documentation of streaming/sending large array via websocket

paccerdk, 2017-02-07:
I am logging temperature data to an array at an interval, and would like to send the data to client via WebSocket, upon connecting.

When doing so, the the library buffers the data, and runs out of free heap quickly.

I could store a copy of the data in a string and send that in a single message, but that would require me to store a duplicate of the data, and take a lot of extra memory.

I can only think of some rather hacky and slow solutions for this, and would like to request documentation, an example, or even just a suggestion for a relatively efficient approach for such a case (using websockets).

Maybe if i could manually flush the buffer, i it should be fairly easy and efficient to do.

I guess this problem applies to all cases where values are being sent at a higher interval than the server transmits, but primarily when dealing with large messages or many messages.

I believe the problem is related to issue #87 

The following is a simplified example and the serial output of it:
[examplews.ino.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/758661/examplews.ino.txt)
[SerialOutput.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/758663/SerialOutput.txt)

me-no-dev, 2017-02-07:
you need to buffer the data ;) that's it, there is no way around it :) I do not know why you are sending text when you can send the buffer in binary form and save lots of time.
In your case, you are abusing the server and api :)

paccerdk, 2017-02-07:
I do the best i can with my current knowledge, i am certainly no expert programmer.
I do however know how to rtfm, but many hours of researching, experimenting, debugging etc. didn't get me much closer to a solution. 
I wasn't aware that binary would be suitable in my case, so i prematurely dismissed it. hence sending text. looking at it now, i can see it makes a lot of sense.

I'll return to buffering and look into sending in binary. Thank you very much! 
(My example works with non-async websocket btw.)

paccerdk, 2017-02-07:
looks like 
`ws.binary(client->id(), logged, logSize);`
does exactly what i was hoping to achieve. (sending my integers to the client without taking up too much memory while doing so)
Thanks again!

me-no-dev, 2017-02-07:
great :) in all cases try to prep as much data as possible into a message, because each message takes memory to be queued for sending and if your messages are many and small, the amount of memory used to hold them would be astronomical :)

me-no-dev, 2017-02-07:
oh and btw your example works with non-async because it's waiting for each message to be sent before returns to you to send the next one ;)

closed 2017-02-07 by me-no-dev

LuckyLuzz, 2018-11-15:
Hey,
did you find a solution to use binary instead of text? To use binary would mean that Json is not suitable for large data?

Regards
Luzz