me-no-dev/ESPAsyncWebServer #127: Support to cancel download or upload when sending a file to client via callback

Matheus-Garbelini, 2017-02-13:
Hello, I have a call back that I send to the client and with the correct headers I can download the file. But if i choose to cancel the download ESP doesn't clear some flag values like `isSendingSD = 0;`
to allow the system send a new download request even if the last one was canceled.
Is there any solution in ESP side or web side that i can recognize a cancel in download? Thanks

``` C++
AsyncWebServerResponse *response = request->beginResponse("text/plain", dataAvaliable,
			[](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
			uint32_t readBytes;
			uint32_t bytes = 0;
			uint32_t avaliableBytes = SDReadFile.available();

			waitSD();

			if (avaliableBytes > maxLen) {
				bytes = SDReadFile.readBytes(buffer, maxLen);
			}
			else {
				bytes = SDReadFile.readBytes(buffer, avaliableBytes);
				isSendingSD = 0;
				SDReadFile.close();
			}

			leaveSD();
			return bytes;
		});

		response->addHeader("Cache-Control", "no-cache");
		response->addHeader("Content-Disposition", "attachment; filename=" + String(name));
		response->addHeader("Access-Control-Allow-Origin", "*");
		request->send(response);
```

me-no-dev, 2017-02-19:
Why not extend a Response class to do the work of sending the file? See how it is done for SPIFFS and do the same. The response object itself knows what to do.

Matheus-Garbelini, 2017-02-23:
@me-no-dev thanks, I actually used a timeout to close somefiles and reset some flags and it worked as well.

closed 2017-02-23 by Matheus-Garbelini