me-no-dev/ESPAsyncWebServer #131: server.serveStatic fails for files larger than about 3K

dagrende, 2017-02-19:
I have made a web server on a WeMos d1 mini, that servers a simple web app.

When I open http://host/morley.csv it works fine (file is 874 bytes). But http://host/box.js does not work (size 8.4K). The server serves only part of the file and get stuck.

The complete app can be found at https://github.com/dagrende/espspiffstest.

Here is the main file espspiffstest.ino:

```javascript
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <Hash.h>
#include <FS.h>
#include "secrets.h"

AsyncWebServer webServer(80);

void setup() {
  Serial.begin(115200);
  delay(200);

  // try connecting to wifi
  Serial.print("\nwifi connecting ");
  WiFi.begin(ssid, password);   // vars set in secrets.h
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.print("\nconnected to: ");
  Serial.println(WiFi.localIP());

  SPIFFS.begin();
  webServer.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");

  webServer.begin();
}

void loop() {
}
```

me-no-dev, 2017-02-19:
Here is an application that I have up and is using the AsyncWebServer and is serving files with size over 1MB. Can you get the raw response packets so we can what is going on?

dagrende, 2017-02-19:
Where is Here :-? How do I best catch the raw packets for you?

me-no-dev, 2017-02-19:
ah sorry, I missed the link :) https://github.com/me-no-dev/esp8266fcgui
tcpdump is a good option for collecting incoming packets from the network. Do a quick google search if you do not know it. It's the most popular tool of the sort :)

dagrende, 2017-02-20:
Thanks!
I started with building your fcgui, to test it. Carefully installing the git versions of the libs. But Arduino IDE 1.6.12 couldn't find ESP8266NetBIOS.h. 

Arduino IDE error message:
Arduino: 1.6.12 (Mac OS X), Board: "Core Development Module, Prebuilt Source (gcc), 160 MHz, 80MHz, DIO, 115200, 4M (3M SPIFFS), ck, Disabled, None"

/Users/dag/work/esp8266fcgui/esp8266fcgui.ino:4:28: fatal error: ESP8266NetBIOS.h: No such file or directory
 #include <ESP8266NetBIOS.h>
                            ^
compilation terminated.

me-no-dev, 2017-02-20:
you did not install the git version of esp8266 arduino

closed 2019-10-05 by stale[bot]