me-no-dev/ESPAsyncWebServer #138: WebSocket messages missed when sent together

cinoan, 2017-02-21:
When connecting to the websocket server and sending multiple messages in the onopen event on the client (Chrome 56 on Galaxy S7 edge) the majority of the time only the first message is registered on the server. very occasionally the first two or three messages are handled. 

All messages are dealt with correctly when using desktop chrome or chrome 56 on a Galaxy S3 but I don't believe it is a client issue. Using the same client and connecting to various websocket echo servers shows everything working fine on the S7 edge.

I have created a basic test server based on the code in the readme
[WSServer.ino.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/789768/WSServer.ino.txt)
and the test client is http://jsfiddle.net/EAVvQ/2673/

The client sends 3 messages in onopen and reports any response it gets, while the server logs each event it handles.

Running that on the S7 egde shows the 3 messages sent but only one reponse (both on the page and in the network frames)
[image]
The server shows just the first message being processed

> WiFi Connected
ws[/ws][1] connect
ws[/ws][1] text-message[%lu]: E1
ws[/ws][1] pong[0]: 

Delaying the sending of the messages, with a 5 second setTimeout in the onopen, results in more responses being received the majority of the time but occasionally just one is handled.

Changing to sending many more messages: -
>websocket.onopen = function() { document.getElementById("output").innerHTML += "<p>CONNECTED</p>";
    for (i = 1; i < 62; i++) { 
        sendMessage('T' + i);    
    }
};

results in a similar issue showing up on the S3 with not all messages being processed
> ws[/ws][125] text-message[%lu]: T1
ws[/ws][125] text-message[%lu]: T2
ws[/ws][125] text-message[%lu]: T3
ws[/ws][125] text-message[%lu]: T4
ws[/ws][125] text-message[%lu]: T5
ws[/ws][125] text-message[%lu]: T6
ws[/ws][125] text-message[%lu]: T7
ws[/ws][125] text-message[%lu]: T8
ws[/ws][125] text-message[%lu]: T9
ws[/ws][125] text-message[%lu]: T47
ws[/ws][125] text-message[%lu]: T48

I wouldn't want to sent that many messages but it looks like there may be some issue with messages getting lost or am I just misunderstanding something?

Thank you

me-no-dev, 2017-02-21:
@sticilface is working on that exact thing :)

sticilface, 2017-02-21:
if ur good with git, try this pull request out. 

https://github.com/me-no-dev/ESPAsyncWebServer/pull/129

Fixes the problem

cinoan, 2017-02-21:
Definitely not good with git but I've had a look and that seems to be just for eventsource rather than websockets. I've tried updating the files and it appears to have made no difference but if it should fix it then that may be my error (I replaced the updated files in arduino>libraries>ESPAsyncWebServer-master and restarted the ide)

Thank you for working on this.

closed 2019-10-05 by stale[bot]