me-no-dev/ESPAsyncWebServer #141: embedd Firmware Upload in AsyncWebserver

kunosee, 2017-03-01:
I have an application based on the ESPAsyncWebserver. Now I want to include the possibility uploading new Firmware through an HTML-Form, like the ESP8266HTTPUpdateServer/Webupdater-Example is showing it.

For that I have written the following Upload-Handler (reduced here to the bare function):
```c++
  server.onFileUpload([](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final){
    Serial.printf("onFileUpload called, index: %d  len: %d  final: %d\n", index, len, final);
    uint32_t maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
    if(0 == index) {
      Serial.printf("UploadStart: %s\n", filename.c_str());
      if(!Update.begin(maxSketchSpace)){ Serial.println("Update begin failure!"); }
    }
    if(Update.write(data, len) != len){
        Update.printError(Serial);
    } else { Serial.printf("Write: %d bytes\n", len); }
    if(final) {
      Serial.printf("UploadEnd: %s (%u)\n", filename.c_str(), index+len);
      if (Update.end(true)) {
        Serial.println("Update succesful!");
       } else {
        Update.printError(Serial);   
       }
    }  
  });
```
To my surprise I see the following output on the serial console:
```
onFileUpload called, index: 0  len: 551  final: 0
UploadStart: TESTFILE.ino.bin
sleep disable
Write: 551 bytes
onFileUpload called, index: 551  len: 1460  final: 0
Write: 1460 bytes
onFileUpload called, index: 2011  len: 1460  final: 0
Write: 1460 bytes
onFileUpload called, index: 3471  len: 1460  final: 0

Panic X:\arduino-1.8.1\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp:98 __yield

```
Like expected, the function is called for each incoming packet. However after the 3rd packet, the core is calling a panic!  What can I do, to avoid this?

me-no-dev, 2017-03-01:
Update class has a setting to set it to async so it does not yield ;)
Look at the header to find it and set it after Update.begin :)

closed 2017-03-01 by me-no-dev

kunosee, 2017-03-01:
Thank you for that hint!
With:
```
Update.runAsync(true);
```
after Update.begin() everything works as expected!

me-no-dev, 2017-03-01:
:) glad you found it