me-no-dev/ESPAsyncWebServer #142: parallel uploads are running in damaged data

wkraft-fablabka, 2017-03-03:
Hello,

I researched some problems related to POST Data transfer and File Upload.  Especially when two requests are coming in at the same time to the same handler (which is most likely the default handler), the packets for those requests are mixed up.
To show the problem I use this default onFileUpload-Function:
```C++
  server.onFileUpload([](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final){
    if(0 == index) {
      Serial.printf("UploadStart: %s  #######################################\n", filename.c_str());
    } else {
      Serial.printf("Upload: %s Index: %d  Len: %d   Final: %d\n", filename.c_str(), index, len, final);
    }
  });
```
and start two parallel uploads:

The result looks like:
```
UploadStart: FileAAAAAA.bin  #######################################
Upload: FileAAAAAA.bin Index: 561  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 2021  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 3481  Len: 1460   Final: 0
... some lines deleted.....
Upload: FileAAAAAA.bin Index: 66261  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 67721  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 69181  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 70641  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 72101  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 73561  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 75021  Len: 1460   Final: 0
UploadStart: FileBBBBBB.bin  #######################################
Upload: FileBBBBBB.bin Index: 555  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 76481  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 77941  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 79401  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 80861  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 2015  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 3475  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 4935  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 6395  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 82321  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 83781  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 85241  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 86701  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 7855  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 9315  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 10775  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 12235  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 88161  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 89621  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 91081  Len: 1460   Final: 0
Upload: FileAAAAAA.bin Index: 92541  Len: 1460   Final: 0
Upload: FileBBBBBB.bin Index: 13695  Len: 1460   Final: 0
...remaining lines deleted
```

This is the same in body data handling, I have choosen the file handler, because there is the filename, which makes debug output more clear.  In the body-handler I have no filename, which can help me to identify.  

For me it looks like, that the function should be a member of the request-class, so that all packets for one request are coming to that unique function.   Furthermore some state-information (like a SPIFFS file handle) need to be tied to that function.

How can that be achieved?

me-no-dev, 2017-03-05:
it can not really :) you are better off using upload handlers per handler. Like server.on("/up1", HTTP_POST, reqHandler, uploadHandler); so you do not mix them up. You could also check if upload is already running and stop the next one.

me-no-dev, 2017-03-05:
scratch that... look at the index/len dumps ;) both uploads are tracked separately ;)

wkraft-fablabka, 2017-03-05:
Thanks fopr your answer!

Did now some more investigation.  Looks like, that I have two nodemcu-boards with defective flash.
That was no damage by conflicting uploads, but caused by unreliable flash.

However one more question:
When I allocate some memory in with the first packet (index == 0), to keep some state information while parsing the incoming data, i can free that memory with the last packet (final == true).

But: How can I free that memory, when the connection times out?  In that case I get no additional call.
Is there a way, to tie additional information to the request object, so that this gets deleted with the timeout?

me-no-dev, 2017-03-05:
You need to extend a Response class for that and free in the destructor. Responses get cleared when connection is closed for one reason or another.

lbernstone, 2018-03-24:
Do you have an example of how to do a server.on with an upload handler?  I would like to have a page with string parameters and file uploads.  I can the parameters just fine, and see the isFile parameters there, but the uploadHandler is never called.

closed 2019-10-05 by stale[bot]