me-no-dev/ESPAsyncWebServer #154: _POST does not work for <input type="number"

kdobrev, 2017-04-04:
I have a form in html5 and all my inputs are  similar to this 
`<input type="number" id="ap" class="semi-square" min="1" max="255">`

These fields does get displayed on the serial monitor. 
```
   server.on("/", HTTP_ANY, [](AsyncWebServerRequest *request){
    int i;
    int params = request->params();
    for(i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isFile()){
        Serial.printf("_FILE[%s]: %s, size: %u\n", p->name().c_str(), p->value().c_str(), p->size());
      } else if(p->isPost()){
        Serial.printf("_POST[%s]: %s\n", p->name().c_str(), p->value().c_str());
```
Any ideas for a workaround?

me-no-dev, 2017-04-04:
yeah... you are missing the input name in html

closed 2017-04-04 by me-no-dev

kdobrev, 2017-04-04:
sorry