me-no-dev/ESPAsyncWebServer #161: Multiple socket connections

zenmanenergy, 2017-04-24:
Is there a way better way to do this? I want to be able to broadcast to multiple sockets, so I'm storing the clients in a global array, like this:

AsyncWebSocketClient * clients[4];

Then on connect I'm doing this:

clients[sizeof(clients)]=_client;

Then to broadcast I'll loop over the array.

But it wants me to specify a storage size. I put 4 in there just to limit it to a random 4 connections, but I'd rather it not arbitrarily limit it. Any suggestions?

me-no-dev, 2017-04-24:
You do not need to track the connected web sockets in order to send stuff to all of them. The WebSocket plugin has methods for that ;)
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.h#L232-L261

zenmanenergy, 2017-04-24:
Awwww yieah! Perfection!

closed 2017-04-24 by zenmanenergy