me-no-dev/ESPAsyncWebServer #163: What am I doing wrong?

zenmanenergy, 2017-04-26:
Can someone point me in the right direction? I'm getting errors with the _ws.onEvent line in the constructor. 

`
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

class GameServer
{
  public:
    GameServer();

  private:
    AsyncWebServer _server(...);
    AsyncWebSocket _ws(...);
    AwsEventHandler onWsServerEvent(AsyncWebSocket * server, AsyncWebSocketClient * _client, AwsEventType type, void * arg, uint8_t *data, size_t len);
};

GameServer::GameServer(){
  AsyncWebServer _server(80);
  AsyncWebSocket _ws("/ws");
  _ws.onEvent(onWsServerEvent);
}

AwsEventHandler GameServer::onWsServerEvent(AsyncWebSocket * server, AsyncWebSocketClient * _client, AwsEventType type, void * arg, uint8_t *data, size_t len){

}
`

closed 2017-05-13 by zenmanenergy

PrintedProto, 2017-06-18:
Were you able to solve this? I'm having the same problem trying to pass an event handler from inside a different class.

zenmanenergy, 2017-06-18:
Nope. I just gave up on it.

PrintedProto, 2017-06-18:
This is the closest I got. Still fails

_ws.onEvent(&AsyncFSWebServer::onWsEvent);

no known conversion for argument 1 from 

'void (AsyncFSWebServer::*)(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, uint8_t*, size_t) 
{aka void (AsyncFSWebServer::*)(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)}' 

to 'AwsEventHandler {aka std::function<void(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>}'

hreintke, 2017-06-18:
@zenmanenergy @PrintedProto 
You should use std::bind(..) when you want to pass a class function.
std::bind(&AsyncFSWebServer::onWsEvent,this, ....placeholders...);

PrintedProto, 2017-06-18:
@hreintke 

Thank you! Your response kinda whooshed over my head, then I looked up info on std::bind and how it works.

http://en.cppreference.com/w/cpp/utility/functional/bind

This line compiled for me. It looks cumbersome though. Is there a way to write it better?

_ws.onEvent(std::bind(&AsyncFSWebServer::onWsEvent,this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3, std::placeholders::_4, std::placeholders::_5, std::placeholders::_6));

hreintke, 2017-06-18:
@PrintedProto 
Agree, it's not the most nice piece of code but that is the way it is.

Not used it myself but I think this should work too.

using namespace std::placeholders;
_ws.onEvent(std::bind(&AsyncFSWebServer::onWsEvent,this, _1, _2, _3, _4, _5, _6));