me-no-dev/ESPAsyncWebServer #164: How to use this library with SD?

rokk0, 2017-04-26:
Im trying to make wifi logger with charts and etc. All content and logs must load from SD card, but default Arduino ESP8266 library is very slow and not asynchronous. I found SD example https://gist.github.com/me-no-dev/01d309cd3760737238cec3468b96f117 but it doesn't work (errors about 'sd::' and 'fs::'). Can someone share simple sketch for SD card hosting with ESPAsyncWebServer please.

Byzula, 2017-05-18:
https://github.com/boblemaire/ESPAsyncSDWebServer

closed 2019-10-05 by stale[bot]

nietaki, 2021-09-28:
For future archeologists:

Wherever the examples serve files from SPIFFS, using `SD` instead of `SPIFFS` in the example code Just Works (tm)!

y3nd, 2022-04-13:
> 

No, it doesn't work well out of the box with big files, only under 4096 bytes