me-no-dev/ESPAsyncWebServer #167: request->authenticate() broken when password is a hash.  Suggested fix.

TheOriginalMrWolf, 2017-05-01:
Hi,

Just trying to use authenticate with a hash instead of plaintext password (ie `request->authenticate(hash)` or `request->authenticate(username, passwordhash, realm, true)`, and login fails every time.  Using `user:realm:md5(user:realm:pass)` as the hash for `request->authenticate(hash)`.

Tracked this down as being an issue with checkDigestAuthentication() in WebAuthentication.cpp.  Specifically, lines 208, 209, 210:
```
  String ha1 = (passwordIsHash) ? String(password) : myUsername + ":" + myRealm + ":" + String(password);
  String ha2 = String(method) + ":" + myUri;
  String response = stringMD5(ha1) + ":" + myNonce + ":" + myNc + ":" + myCnonce + ":" + myQop + ":" + stringMD5(ha2);
```
It seems that regardless of whether the password is hashed or not, ha1 gets hashed again in line 210.  This means the hash of the hash is used, instead of the hash.  In other words, if the password is not hashed, it uses md5(user:realm:password), but if the password is hashed (ie already md5(user:realm:password)), it then uses md5(md5(user:realm:password)).

Suggested fix (works for me):
```
  String ha1 = (passwordIsHash) ? String(password) : stringMD5(myUsername + ":" + myRealm + ":" + String(password));
  String ha2 = String(method) + ":" + myUri;
  String response = ha1 + ":" + myNonce + ":" + myNc + ":" + myCnonce + ":" + myQop + ":" + stringMD5(ha2);
```
This way response uses md5(user:realm:password) regardless of whether the password is hashed or not.

What do you think?

TheOriginalMrWolf, 2017-05-03:
*bump*.... sorry...

TheOriginalMrWolf, 2017-05-31:
non?

me-no-dev, 2017-05-31:
seems I totally missed this issue and the bump? how could that happen?
BTW another user reported this same thing. Why don't you make a PR?

milkpirate, 2017-06-13:
@TheOriginalMrWolf thank you for your comment! your solution works like a charm. will close my issue.

omersiar, 2017-06-21:
@TheOriginalMrWolf @milkpirate Do you have HTML or Pure Javascript example?

milkpirate, 2017-06-21:
@omersiar dont know what you mean. Its all server side. A minimal working example would be:

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <WebAuthentication.h> // otherwise the IDE wont find generateDigestHash()

AsyncWebServer server(80);

#define HTTP_USER     "admin"
#define HTTP_PASS     "1234"
#define HTTP_REALM    "realm"

void setup() {
  Serial.begin(74880);
  
  WiFi.mode(WIFI_AP_STA);       // connect to wifi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  server.onNotFound([](AsyncWebServerRequest *request) {
    if(!request->authenticate(generateDigestHash(HTTP_USER, HTTP_PASS, HTTP_REALM).c_str())) {
      return request->requestAuthentication(HTTP_REALM, true);
    }
    request->send(200, "text/plain", "login ok");   // <-- the only thing that might be 
                                                    // considered HTML is the "login ok"
  });

  server.begin();
}

void loop() {

}
```

omersiar, 2017-06-21:
@milkpirate Thank you for your kind response. But this will open a Auth Popup on Chrome (Browser).

I want to send response from Browser to ESP via Javascript. 

http://www.peej.co.uk/articles/http-auth-with-html-forms.html

But there is no example to send such response with Javascript.

milkpirate, 2017-06-21:
No at least I dont have such an example. You should try stackoverflow or so and ask there. Here isnt the best place to do so because is not an issue with the server running on the ESP.

TheOriginalMrWolf, 2017-06-21:
First result on google for "javascript digest authentication": https://github.com/inorganik/digest-auth-request

Generally good form to try this yourself first...

omersiar, 2017-06-21:
Yes i have tried everything i found so far. I thought maybe you guys have a simple solution. Thank you.

milkpirate, 2017-06-21:
@omersiar does the sketch I posted work with your setup?

omersiar, 2017-06-21:
@milkpirate I do not understand how it works. I'm trying to learning, so far I put exact code from your post to ESP.

Then as usual I try to navigate a non-exist page via Chrome Browser. Then Chrome pops up for username and password. Should it? I do not know.

I never get "login ok" from ESP.

How do you send response from browser to esp?

TheOriginalMrWolf, 2017-06-21:
Yes, the popup should come up when you call up a page on the ESP that ends up calling the code mentioned.  You enter your username and password into the chrome popup...

If you don't want a popup, then on the client side use the javascript library I pointed you at.

milkpirate, 2017-06-21:
Well you have to make changes to files of the ESPAsyncWebServer library as TheOriginalMrWolf in the OP suggested. Otherwise it wont work. Neither with the browser nor with the JavaScript. Do so and test my Sketch again with user: admin and pass: 1234. If your broser then shows you "login ok" you can go on trying to fiddle with JavaScript.

TheOriginalMrWolf, 2017-06-21:
Also, since me-no-dev just committed an update to the lib which incorporates my fix, you can just re-clone/download the library...

milkpirate, 2017-06-21:
Hey guys as all the attention is right on this issue I found something new not working. With the above sketch everything works so far. I would like to make a get call and transmit some parameters. The usual way works

```http://[IP]/file.html?abc=123&def=456```

but then I would like to post a JSON string like so:

```http://[IP]/file.html?abc={"key1":"value","key2":0}```

and the authetification fails (like before the latest fixes). I narrowed it down to the use of a `,` in the parameter string.

```http://[IP]/file.html          <-- auth works```
```http://[IP]/file.html?,      <-- doesnt```

Any idea why?

me-no-dev, 2017-06-22:
your query string has to be url encoded ;)

milkpirate, 2017-06-22:
Stupid me...

closed 2019-10-05 by stale[bot]

mike-s123, 2020-02-23:
Just asking why commas affect authentication. Sure, they're HTML reserved characters, but so are colons, which don't need to be escaped for ESPAsyncWebServer. Also, commas work just fine in the "simple" ESP web servers (ESP8266WebServer.h and WebServer.h).

    /rest?json={"addr":0%2C"cmd":"read"%2C"count":2}

works

    /rest?json={"addr":0,"cmd":"read","count":2}

doesn't. Why?