me-no-dev/ESPAsyncWebServer #168: Other way to handle longtime tasks on the server side as to poll it

pit001, 2017-05-02:
Hi, thank you for your great implementation. I read your documentation and found the "Scanning for available WiFi Networks" example. Starts scanning and polls the result (as I unterstand).
Is there a other way to do this - without polling ?

-> The client requests (by using ajax) data from the server (ESP w AsyncTCP), the server processes this data (>5sec) and after that it send it back to the client as an asyn responce (Ajax style). 

thank you in advice
 Peter

me-no-dev, 2017-05-02:
Appart from taking the request outside of the server, the other options are WebSocket and EventSource

omersiar, 2017-08-14:
Updated Core has this feature, it may help.

http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/scan-class.html#scannetworksasync

closed 2019-10-05 by stale[bot]