me-no-dev/ESPAsyncWebServer #169: wyhy

c-mysec, 2017-05-05:


me-no-dev, 2017-05-06:
there are very few things that can be done and it will not be dramatic as result either ;) he regular server uses about the same memory also when parsing. Just does it linear and at the end it frees it. For you the query string is 41-500 bytes, but the reality is different :) there are headers and other data also. When you parse a form specifically, it's quite intensive.

c-mysec, 2017-05-07:
In my tests, I only added form params. Headers remain the same for all requests. But memory usage increase rapidly with new parameters.

c-mysec, 2017-07-06:
Is there a way to make AsyncWebServer delay request parsing momentarily while I am doing memory intensive processing in the loop? something like server.pause() and server.continue()?

closed 2019-10-05 by stale[bot]