me-no-dev/ESPAsyncWebServer #170: beginChunkedResponse won't tell when client disconnected

aguaviva, 2017-05-18:
In the following code, how do I know when the client disconnected? Apparently there is no way to know that.
```
AsyncWebServerResponse *response = request->beginChunkedResponse("application/octet-stream", [&response](uint8_t *buffer, size_t maxLen, size_t index) -> size_t
{
    memcpy((char*)buffer, "hola",4);
    return 4;
});
response->addHeader("Server","ESP Async Web Server");
request->send(response);```

me-no-dev, 2017-05-19:
once you send the data, then next time that the function is called you need to return 0 to indicate that you will not be sending more data. Index will show you how much you have sent already.

closed 2017-05-19 by me-no-dev

aguaviva, 2017-05-19:
right, that is if the server wants to stop the transmission, but in the case where the client disconnects, how do you know that in that function? is there any way to check for that?

me-no-dev, 2017-05-19:
the client will be disconnected by the server after you return 0 in that callback. It's done asynchronously and out of your control :)

aguaviva, 2017-05-19:
I understood that case, the case where the server wants to finish sending data. This is a different case, in this one the client closes the connection when I am half way sending data. I need to know when that happens so I can release some memory I need to allocate. Thanks!

me-no-dev, 2017-05-19:
uhm... request->client()->connected() maybe?

aguaviva, 2017-05-19:
I just tried that and it didn't work, once the client closes the connection the callback is never called again.
Being told when the client closes the connection is important so the server can release the resources allocated to that connection. Actually I am using your lib for streaming audio to a browser, once it works I'll put the whole project on GitHub.

Thanks for this fantastic library BTW :)