me-no-dev/ESPAsyncWebServer #171: Need clarification on HTTP_POST body handling

faceless2, 2017-05-19:
Hi there

Not sure if this is an issue with the code or my understanding. I am trying to create a handler for a POST event which received raw data (JSON, or similar, rather than form-encoded content).

I've written my handler like this:
```
webserver.on("/write", HTTP_POST, [](AsyncWebServerRequest *request) {
        Serial.println("write request handler");
    }, NULL, [](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) {
        Serial.println("write body handler");
  });
```
which looks to be correct based on the "on" function signature. But the "body handler" function is never called if I POST to this method with curl - just the "request handler".

Am I missing something? Is this not the correct way to do it? I can find no examples of reading a POST method beyond the one on the main documentation, which is set up as a catch-all handler rather than applied to a single handler.

faceless2, 2017-05-22:
To follow myself up: the syntax above is correct, but the "body" handler is not called if the content-type is "application/x-www-form-urlencoded". Might be worth noting this in the documentation.

closed 2017-05-22 by faceless2

me-no-dev, 2017-05-22:
urlencoded posts go to the post variables. That is what is expected by sending the urlencoded type :)

FarshidAhmadi, 2021-05-25:
i am using aboave code but nothing happen in serial monitor when i press Send button  in postman .also the body is Hello Word (not empty)