me-no-dev/ESPAsyncWebServer #172: Websocket doesn't work and Internet Explorer version

drony, 2017-05-21:
Websocket doesn't work and Internet Explorer version.

MoegerleStephan, 2017-05-28:
I'm sadly having the same issue.
Could this have something to do with the security configuration for the intranet of the IE?

me-no-dev, 2017-05-29:
nope. It has to do with the fact the MS does not follow the rules for constructing requests. The server expects the "Host" header to be the first header and that is not the case with IE, so it breaks the parser

MoegerleStephan, 2017-05-29:
Ok.. that's exactly the problem..
I've tried to log the request..

Google Chrome and Firefox send the request as a whole (inclusive HEADER), what fits perfectly in your parser..
> NOT_FOUND: GET http://192.168.1.1/ws
_HEADER[Connection]: Upgrade
_HEADER[Upgrade]: websocket
_HEADER[Origin]: http://www.websocket.org
_HEADER[Sec-WebSocket-Version]: 13
_HEADER[Sec-WebSocket-Key]: cbm715DQGb5Qd9eyI2nZiw==
_GET[encoding]: text

In IE it looks more like following: (All Header files were transmitted seperatly)
> NOT_FOUND: GET http://192.168.1.1/ws
_GET[encoding]: text

Did you already see the parser implementation of Links2004 ? 
https://github.com/Links2004/arduinoWebSockets/blob/master/src/WebSocketsServer.cpp#L676

This parser works perfectly with the newest version of IE..

me-no-dev, 2017-05-29:
it's one thing to be a WebSocket server with the sole purpose of parsing WebSocket requests and a whole other thing to be an actual web server parsing all kinds of requests and supporting all kinds of things.

MoegerleStephan, 2017-05-29:
I see.. Sorry I didn't wanted to be disparging..
You did a great job with your async webServer. I'm excited day by day of your great work!

I will try to merge the idea of Links2004 into yours and come back in the next few days..

me-no-dev, 2017-05-29:
It needs a different approach to get this resolved. But that will mean even more memory usage as you need to collect all headers before actin on them. The Host header in question splits the logic to virtual hosts and together with the request can make a decision if a given request can be handled or not.

universam1, 2017-06-30:
Could we for the time being introduce a IE compatible directive for the ones desperately needing it?

pdfelen, 2017-07-31:
I am working on a solution for this. It is caused by 2 separate small bugs. I will post the solution next week. I don't have a good Internet connection at hand this week... 
Bug 1: websocket needs Host in header and this implementation assumes this header is the first to be transmitted by the browser. IE and Edge send it as one of the last.
Bug 2: if a handler exists that results in a file or folder that has the same name as the folder in the socket request, it will reply with result code 200 and transmit the file. While the browser expects code 101.

pdfelen, 2017-08-06:
On implementation I had 2 choices:
1- Parse headers twice (first to determine request type and Host, attach correct handler, second to store needed headers)
2- Parse headers first, attach correct handler, remove not needed headers
I have chosen to implement the second one, it temporary uses more memory, but not needed part will be freed fast

pdfelen, 2017-08-07:
The uploaded fix contains my proposed fixes to issues 172 and 198
[fixed issues 172 + 198.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1203563/fixed.issues.172.198.zip)

me-no-dev, 2017-08-18:
Merged into master :)

closed 2017-08-18 by me-no-dev

pdfelen, 2017-08-18:
great job!

universam1, 2017-08-18:
thanks @pdfelen and @me-no-dev