me-no-dev/ESPAsyncWebServer #175: Having trouble with digest authentication

milkpirate, 2017-05-29:
I am having trouble with the digest authentication. Well... its not accepting the password... or user... dont know. Code is pretty much like below (using Chrome). Is there anything I am missing?

```
#include <WebAuthentication.h> // otherwise the IDE wont find generateDigestHash()

#define HTTP_USER     "admin"
#define HTTP_PASS     "1234"
#define HTTP_REALM    "realm"
...

server.onNotFound([](AsyncWebServerRequest *request) {
	Serial<<generateDigestHash(HTTP_USER, HTTP_PASS, HTTP_REALM)<<endl;  //	admin:realm:b0e2bac38418dc35a829b615e73f96d6

	if(!request->authenticate(generateDigestHash(HTTP_USER, HTTP_PASS, HTTP_REALM).c_str())) {
		return request->requestAuthentication(HTTP_REALM, true);
	}
	request->send(200, "text/plain", "login ok");
});

```

me-no-dev, 2017-05-29:
hm... it looks fine. Have you tried to auth directly and not use premade hash?

milkpirate, 2017-05-29:
No. Doesnt work either. Heres the negotiation with `generateDigestHash()`:
```
General
Request URL:http://192.168.0.43/
Request Method:GET
Status Code:401 Unauthorized
Remote Address:192.168.0.43:80
Referrer Policy:no-referrer-when-downgrade

Response Headers
Accept-Ranges:none
Connection:close
Content-Length:0
WWW-Authenticate:Digest realm="realm", qop="auth", nonce="5a09491d308b74a980e575703175824e", opaque="8b5d8b12b9e74ed6b13a791c94e9244a"

Request Headers
Accept:text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
Accept-Encoding:gzip, deflate, sdch
Accept-Language:en-US,en;q=0.8
Authorization:Digest username="admin", realm="realm", nonce="f76cc6c213764609dda82bf363106e7b", uri="/", response="f2ff8c89aeb86b788511c4c8a0d90865", opaque="46ce184990351d05a2277c3b907968de", qop=auth, nc=00000001, cnonce="8d3dc399976c99a4"
Cache-Control:max-age=0
Connection:keep-alive
Host:192.168.0.43
Upgrade-Insecure-Requests:1
User-Agent:Mozilla/5.0 (Windows NT 6.1; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/58.0.3050.0 Iron Safari/537.36
X-CookiesOK:I explicitly accept all cookies
```

me-no-dev, 2017-05-29:
what is the next response? what I see is the first request, then asked for password and then browser calling back with the AUTH set.

milkpirate, 2017-05-29:
Thats all what chrome is showing me:

[image]
[image]

TheOriginalMrWolf, 2017-06-13:
See https://github.com/me-no-dev/ESPAsyncWebServer/issues/167

Should help :)

TheOriginalMrWolf, 2017-06-13:
Sorry I haven't raised a pull request for my fix yet - never done that before so gotta work out how to do it (in all my spare time...)!

:)

milkpirate, 2017-06-13:
@TheOriginalMrWolf's solution works. See [#167](https://github.com/me-no-dev/ESPAsyncWebServer/issues/167).

closed 2017-06-13 by milkpirate