me-no-dev/ESPAsyncWebServer #177: serving font-awesome or font glyphicons

micha1983, 2017-06-15:
HI,
i try to server font files from spiff. but when i serve them locally from the ESP8266 i get an 404 error.

Is anybody able to server woff, tff or woff2 fonts with this project and can help me?

I included the css file inside of the index.html like this

`<link rel="stylesheet" href="css/font-awesome.min.css">`

and i get following 

`GET http://192.168.16.124/fonts/fontawesome-webfont.woff2?v=4.7.0 
192.168.16.124/:1 Failed to decode downloaded font: http://192.168.16.124/fonts/fontawesome-webfont.woff?v=4.7.0
192.168.16.124/:1 Failed to decode downloaded font: http://192.168.16.124/fonts/fontawesome-webfont.ttf?v=4.7.0
192.168.16.124/:1 Failed to decode downloaded font: http://192.168.16.124/fonts/fontawesome-webfont.woff?v=4.7.0
192.168.16.124/:1 Failed to decode downloaded font: http://192.168.16.124/fonts/fontawesome-webfont.ttf?v=4.7.0`

Can anybody help me out. sorry i'm pretty new to this stuff.
maybe there is something principal wrong.

nuljon, 2017-06-16:
Try leading with a slash in your url ...
href="/css/font-awesome.min.css

  (#177)

micha1983, 2017-06-16:
a slash in front doesn't change anything.
it's only with fonts. alle other files will be served nicely.

micha1983, 2017-06-16:
If I implement this with a path from a different server it works fine. so something is wrong in serving this files with ESPAsyncWebserver

micha1983, 2017-06-16:
i found it.

the pathname is too long. i removed everywhere -webfont and than it worked.

closed 2019-10-05 by stale[bot]