me-no-dev/ESPAsyncWebServer #179: Size Limits with Responses

romanseidl, 2017-06-18:
This might be a stupid question but when I do the following:

```
void handleSizeTest(AsyncWebServerRequest *request) {
  AsyncResponseStream *response = request->beginResponseStream("text/plain");
  for(uint16_t i = 0; i < 500; i++){
    response->printf("%10u",i);
  }
  request->send(response);
}
```

I keep getting 500 Numbers (=5000 chars). Is there some way to avoid this limit? I am having trouble serving larger pages.

romanseidl, 2017-06-19:
I can't find a hint on why this is happening. Maybe it is just a memory limit of some kind, but I can't fint it hardcoded, which would be very stange as 5000 chars is somehow a very distinct number. I will try in a clean environment with more heap space available, maybe the number changes.

Maybe one should think of sending a chunked response in case one wants to send a large web page? It should be possoble to implement a Print interface (like the AsyncResponseStream) that does not collect all responses im memory but flushes to the response stream as a chunked response.

Has anybody tried that?

romanseidl, 2017-06-21:
I now tried with a clean environment and the result is just the same.

One can find a complete test sketch on: https://gist.github.com/romanseidl/c8c5606c7138b9227fecc20aeb17f850

Is this the expected behaviour? And if so is it possible to change it?

romanseidl, 2017-06-21:
It's obviously a matter of free memory, which is not such a surprise as the AsyncResponseStream ist just a buffer in the heap.

I produced a memory leaking test script which returns less and less characters:
https://gist.github.com/romanseidl/d2b51cbbef55a062c5b1d07bf547561e

So there could be some solution which e.g. uses chunked responses?

romanseidl, 2017-06-21:
I think my idea on sending chunked responses is somehow impossible due to the design of the async server?

me-no-dev, 2017-06-21:
OK, so the issue is that response stream can only allocate the space currently available for sending. That could be one packet or two (1460/2920). If you want to send large data, chunked encoding is your friend. Also you can use StreamString to stream all data to a string and then send that, which in turn will chink it in packets and send out.

romanseidl, 2017-06-21:
Thanks for the reply.

As far as I can see StreamString is actually doing the same as AsyncResponseStream, it just collccts the response in memory. It uses a String whicle AsyncResponseStream uses a cbuf but thats about it.

I could send large amounts of numbers via chunked encoding, thats right. 

But in the end the number case was just a stupid example. In the real world a web page is commonly generated in spaghetti code (aka templates). And then is would be kind of tedious to name chunks and thus several reentry points that create different parts of the page. And as I now understand the server, it pulls the chunked response object for chunks of data because it needs them on client request for more. It's event driven and waits for the client ack.

I would be nice to just ignore that, write to a buffer line by line in spaghetti code while the buffer gets emptied on the other side. 

But as I understand that would need for a concurrent operation - something would have to empty the buffer and the writes would have to block until the buffer gets emptied before continuing the writes. And then still one would have the chance of even bigger heap problems as several write operations beeing open at the same time that may produce arbitrary artifacts on the heap (aka model) that may be much larger than the page result. So for real no other requests can get served until one request is finished writing. 

As far as I see the only viable solution for that scenario is a sycronous or blocking web server. Which is maybe something which contradicts the basic idea of this project.

me-no-dev, 2017-06-21:
The Request object itself has a couple of placeholders. One for a file and one that can be any pointer to anything. So you can allocate a buffer and use it, then free at the end and that buffer will be attached only to this request (you can handle others).
Generally there is a way around everything ;) but it's rarely obvious with Async.

romanseidl, 2017-06-21:
> "can be any pointer to anything"
Yes, but there must be something. Either a method or a result ready to be served. 

> So you can allocate a buffer and use it, then free at the end and that buffer will be attached only to this request (you can handle others).
Sure but as far as I understand then I need a complete page in the buffer. 

I don't see a way where I could wrtite into a buffer in some kind of spaghetti code an block the process when the buffer is filled to a certain extent until it gets emptied in the background. Or am I missing something?

me-no-dev, 2017-06-21:
why do you want to block it though? that is what I do not understand

me-no-dev, 2017-06-21:
maybe if you give me a more exact explanation of what you are trying to do, I might get some idea. Where is the data coming from? Is it coming in chinks? What size?

romanseidl, 2017-06-21:
I don't know if I have to block. There is the following structure:

`[PageMethod] -> [Buffer] <- [Webserver / Response] <-> [Client]`

In general I see three options how the "PageMethod" and the buffer may interact:

1. **Write the whole page** into the buffer
  this is what happens with the AsyncResponseStream, but then I run out of heap.
2. **Split up the page** into parts that fill seperate buffers
  as I understand this should be possible with the chunked response, **but** this needs extra work to explicitly split the page
3. **Block the pagewriting** in case the buffer is full until it has been emptied or kill it on timeout
   I don't see how this should be possible at the monent. If I block a process it makes the esp kill itself.

_Now for an example:_ I have a page that is a log viewer. 

I serve it by setting up a model (opening the log file) and then including a header template, looping through the log lines and finally including a footer. Actually I do this by using plain #includes, but a methods would also do (with some overhead though).

I can only show a very limited amount of lines in that log viewer else the output gets truncated as the buffers fill up my heap. And this is even though I have 30k of heap available which is not really too bad for an esp.

romanseidl, 2017-06-22:
As an additional example: I wrote a buffer that does this with the simple sync server. But when compared to the async server it's dead slow:
https://github.com/esp8266/Arduino/issues/3366

me-no-dev, 2017-06-22:
Here is an idea. You can extend a Response class to generate the page and manage the buffer. Then just send that Response to the request. It will give you a fine grain access to what you might need and a destruct method to clean everything at the end.

me-no-dev, 2017-06-22:
Each response will be on it's own attached to a given request, so you can still do multiple at the same time.

romanseidl, 2017-06-22:
Hm, this may work if I save the call context if the buffer is full, then return the buffer to the server and then restore the call context on the next call for data. Seems very low level but maybe this could work.

I am still not sure if you get my point about the stupid spahetti code (aka templates) producing the page. I don't want to know anything about his code's structure, I just want to stop it where it filled up the buffer and make it continue writing to the buffer when the buffer is empty again.

Does the server take care of killing request-objects that never get an ack response?

me-no-dev, 2017-06-22:
ACK is part of TCP and yes ;) the AsyncTCP takes care of stuff.
Response is your friend ;) it is lower level and harder to write, but that is why the benefits are so big at the end.

romanseidl, 2017-06-22:
Thanks, so at least that`s not the issue here.

Still I don't have an idea if and how one could save/restore a call context on an esp.

romanseidl, 2017-06-22:
As far as I now understand I would at leas need something like coroutines support for this to work. Are there even any possibilities to do this in the esp-arduino environment?

I am not too fast at figuring out something like this I am more an expert in java or scala or python than in c and c++. But still I like it, probably because it makes me feel young again.

jej, 2017-06-24:
I think I am facing the same problem... I have this response:
```
	server.on("/config", HTTP_GET,
			[](AsyncWebServerRequest * request) {

		body = "\
    <h1>Configuration</h1>\
    <form class=\"form-horizontal\" method=\"POST\" action=\"/config\" enctype=\"multipart/form-data\">\
      <div class=\"form-group\">\
        <label for=\"veille-temp\" class=\"col-sm-4 control-label\">t° seuil veille</label>\
        <div class=\"col-sm-4\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"veille-temp\" value=\"" + String(veilleTemp) + "\">\
            <div class=\"input-group-addon\">°C</div>\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <label for=\"temp-veille-periode\" class=\"col-sm-4 control-label\">Période lecture t° veille</label>\
        <div class=\"col-sm-4\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"temp-veille-periode\" value=\"" + String(tempVeilleReadPeriodeS) + "\">\
            <div class=\"input-group-addon\">s</div>\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <label for=\"vigilence-temp\" class=\"col-sm-4 control-label\">t° seuil vigilence</label>\
        <div class=\"col-sm-4\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"vigilence-temp\" value=\"" + String(vigilenceTemp) + "\">\
            <div class=\"input-group-addon\">°C</div>\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <label for=\"temp-vigilence-periode\" class=\"col-sm-4 control-label\">Période lecture t° vigilence</label>\
        <div class=\"col-sm-4\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"temp-vigilence-periode\" value=\"" + String(tempVigilenceReadPeriodeS) + "\">\
            <div class=\"input-group-addon\">s</div>\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <label for=\"temp-periode\" class=\"col-sm-4 control-label\">Période lecture t°</label>\
        <div class=\"col-sm-4\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"temp-periode\" value=\"" + String(tempReadPeriodeS) + "\">\
            <div class=\"input-group-addon\">s</div>\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <label for=\"max-temp\" class=\"col-sm-4 control-label\">t° max. cible</label>\
        <div class=\"col-sm-4\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"max-temp\" value=\"" + String(maxTemp) + "\">\
            <div class=\"input-group-addon\">°C</div>\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <label for=\"date\" class=\"col-sm-4 control-label\">Date</label>\
        <div class=\"col-sm-2\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"date\" placeholder=\"Jun 21 2017\" value=\"\">\
          </div>\
        </div>\
        <label for=\"time\" class=\"col-sm-1 control-label\">Heure</label>\
        <div class=\"col-sm-2\">\
          <div class=\"input-group\">\
            <input type=\"text\" class=\"form-control\" name=\"time\" placeholder=\"23:49:25\" value=\"\">\
          </div>\
        </div>\
      </div>\
      <div class=\"form-group\">\
        <div class=\"col-sm-offset-4 col-sm-8\">\
          <button type=\"submit\" name=\"save\" class=\"btn btn-success\">Enregistrer</button>\
          <button type=\"submit\" name=\"home\" class=\"btn btn-default\">Retour</button>\
        </div>\
      </div>\
    </form>";
				request->send(200, "text/html", htmlHeader + body + htmlFooter);
			});
```
It works in that case, but if I add a new "form-group" div the response is blank... Probably it exceeds the heap size. The page size is 3839 bytes.
Well, I think I need to downgrade to ESP8266WebServer...

romanseidl, 2017-06-24:
I went back to using the sync server and this scheme: 
https://gist.github.com/romanseidl/4d6ac8c247a4768417037b1a00b3bb9f
This allows for very long responses, as long as they come in blocks of reasonable size. I tested it with up to 1MB.

Actually I think there is no solution to my problem as I think that there is nothing like coroutines in ESP Arduino. But I might be wrong an would still like to get to know some solution. There was some people who developed some macros that emulated coroutines, maybe this could work?

There is so many problems to simple web development. The other one is, that you can't use multiline raw String with the "F()" macro because esp arduino uses a dead old gcc for no known reason that has a bug with multiline macros.

jej, 2017-06-24:
Thanks for your feedback and suggestion.
I don't really need a "very long response", but 4kB is quite small these days for a web page.
Maybe I did a mistake in my code. If not, there is a kind of incoherence. My sketch is not so big (30% of max size) and actually does nothing except managing some settings through a webpage. ESPAsyncWebServer implements HTTP serving very well, but if its size and memory consumption allow only to serve basic pages, that's a kind of dead lock situation...
Hope it's on my side!

```
Le croquis utilise 343221 octets (32%) de l'espace de stockage de programmes. Le maximum est de 1044464 octets.
Les variables globales utilisent 44704 octets (54%) de mémoire dynamique, ce qui laisse 37216 octets pour les variables locales. Le maximum est de 81920 octets.
```

romanseidl, 2017-06-24:
If you want to know how much free space there is at runtime you need to call ESP.getFreeHeap().

I don't think that the async server is using the stack but I'm not sure. Anyway I also had problems with larger pages, also meaning pages of maybe 4kb or something.

The only way for the moment I know of on how to solve this with the async sever is to split your page up into distinct parts (i.e. methods?) that are produce of the page parts one after another in a chunked response.

jej, 2017-06-24:
Well, I just replaced with ESP8266WebServer, working well without so much changes.
I don't want to have to hack things just to server a 4kB page at the speed of light!

me-no-dev, 2017-06-25:
@jej I can totally async your case ;)
I would create a new response class and will initialize it with an array of the things that you want to send (strings in your case), then while sending I'll keep track on which part em I and how far I have gotten, then send as much as I can and start a new part (even in the same buffer). This would allow simple "templates" like this to run async.
@romanseidl some sort of similar thing could also be applied to your case. Know what you need to send and how much you have send, then keep a buffer for what you generate (and have left for next time).
Surely you are free to use the Sync server as well :) Keep in mind that I wrote it much earlier than the async one and that it lacks quite a bit of features compared ;)

romanseidl, 2017-06-25:
> Some sort of similar thing could also be applied to your case. Know what you need to send and how much you have send, then keep a buffer for what you generate (and have left for next time).

This is exactly what I don't *want to know* in advance. And I suspect that @jej doesn't want to know either. Actually nobody wants to know that if it can be avoided. Because then you have to think in advance about every method producing a larger web page how to split into parts small enough to fit into memory, etc. 

I don't love the sync server and yes it lacks some features. But I don't have any Idea on how to implement sending large pages without splitting the code with the async server. Else one would need something like coroutines and that's AFAIK nothing that would b possible with ESP Arduino.

me-no-dev, 2017-06-25:
even the regular server spits everything in packets in the background ;) but as you know that can be done only connection by connection. If you don't want to bother to go through the hurdle of thinking Async, then better use the sync servers ;) I totally admit that Async is for way more complicated applications than just sending some page.

romanseidl, 2017-06-25:
> even the regular server spits everything in packets in the background ;)
In the background is what I want!

The problem is not *"thinking async"*, it's not that I am too lazy, it's that I don't want to produce more code and complexity than necessary. Splitting pages produces enormous overhead on web development.  As I already wrote some time ago web development is commonly producing spaghetti code. This is common practice and quite the opposite of "thinking async". It's quite the opposite of thinking about anything else but the page code. It's not a surprise that web frameworks on large scale machines never bother you with details about concurrency, buffers, package size etc. It's just handled for you. 

With what is the current state of the async server one maybe lacks patterns or tools that facilitate how to split up a page into chunks of smaller sizes. I would have to think about that. A generic "no brains" solution could only be done when there is something like coroutines.

me-no-dev, 2017-06-25:
then the sync server is for you ;) I feel this conversation going nowhere :D I will not make the server synchronous ;) and since you do not want to make your app more complex, this discussion should be over

closed 2017-06-25 by me-no-dev

romanseidl, 2017-06-25:
Well, what I find a bit frustrating is that you seem to think that this is *my* problem. It isn't really. As for me I don't care which server I am using or whatsoever. 

I wanted to solve a generic problem and I still have the feeling that you are not grasping it. At least you never seemed to really understand it. All answers always come down to "then this is nothing for you" and then somehow the underlying accusation of me beeing to lazy to thin async. If I were I wouldn't try. And because of this there is no end to this discussion. We are talking about different things. 

This is not what I like to invest my time in. Actually I would love to learn but also make other people get new insight. I have been a developer for 25 years and I have been developing web frameworks. It's not that I wouldn't have a faint idea about what's going on out there. 

I would have at least hoped to make you to think about this problem, that's all. It seems it can't be solved for now anyway, maybe it can in the future. As I would hope for you being in the IOT-industry system for a little while you might one day come up with a better solution than just telling people to just think async.

me-no-dev, 2017-06-25:
I mean no disrespect here, but as you prompted coroutines are not supported, so you can not interrupt a task a continue from then on. At the same time I tried to give you an advice at what you should look at for inspiration for solution and how I would have gone trying to overcome this problem. You have to understand that not all uses can be covered by everything. Some solutions need to be fully custom. I understand you have a problem, but at the same time I can not help you further than what I have already done. This server works with data in chunks and expects you to provide it with the necessary amounts when called for. That is how the file response and others work. You need to buffer your own way if you will overflow the given buffer and manage this yourself. This is not a computer and we do not have unlimited amounts of memory, therefore we as library developers try to use as little as possible and give the users enough for their actual application and other libs.
A simple template Response is not a big deal to make, but then it will not cover everyone's needs (yours included).

jej, 2017-06-26:
Whaooo, it's war here now :)
Not sure to understand everything about the HTTP layer concern here, in general I deal with the layers above... In fact, like @romanseidl I don't want to burn my brain in TCP/HTTP just to send 4kB :)
Now I use sync webserver, but it hangs just when I Ctrl+R(eload) page, probably because all css/js requests are sent concurrently or I don't know what... When files are cached it's ok, but full reload of static files hangs. Dead lock again... Pitiful life :(
So I could think about going back to async because in the past it was a good deal to solve another problem I had, and I really thing it implements formally the question.
I agree to split my response myself if needed (even if I don't understand why this "million dollars library" cannot do it for me if it's so simple...). Is there any doc that explains how to send multiple blocks to answer one request? I guess I cannot just loop on `request->send(200...`?
Well, now I understand a bit why IoT frameworks put all the things in cloud rather than trying to make a local solution... I need local solution.
Thanks anyway for your time and everything else.

jej, 2017-06-26:
Here maybe? #113

hreintke, 2017-06-26:
Is there any reason not to use a SPIFFS file (or part of the flash) as the "buffer".

romanseidl, 2017-06-26:
> You have to understand that not all uses can be covered by everything. Some solutions need to be fully custom.

That can't be avoided, even more so on such a small system. Still one should strive to allow for maximum flexibility and an easy to use system.

> This is not a computer and we do not have unlimited amounts of memory, therefore we as library developers try to use as little as possible and give the users enough for their actual application and other libs.

Well I would doubt it's technically speaking not a computer. I might be dead old but I remember those big machines on the desktop used to be limited in a similar way. Anyway you are right one has to be very thoughtful on what you provide as a general infrastructure then because it eats up space.

> Whaooo, it's war here now :)

No, I think not quite yet. At least I hope so.

> Now I use sync webserver, but it hangs just when I Ctrl+R(eload) page, probably because all css/js requests are sent concurrently or I don't know what... 

That's strange, I never had any "blocking" problems. It just becomes very slow as the browser / client hat to wait for quite a while.

> Is there any doc that explains how to send multiple blocks to answer one request? I guess I cannot just loop on request->send(200...

You should need to define a chunked response. The docs are quite sparse but maybe there is some good examples?

> Is there any reason not to use a SPIFFS file (or part of the flash) as the "buffer".

Speed and flash lifetime I suppose. But I am not sure how slow it would be and how fast it would kill the flash memory.

jej, 2017-06-26:
@hreintke I did that way because I set some values in the fields (`<input type=\"text\" class=\"form-control\" name=\"veille-temp\" value=\"" + String(veilleTemp) + "\">\`) There are several ways to do that or not to do at all (empty fields, modified if a value exists when POST, or jQuery to fetch values, etc), but at first I didn't except to be annoyed by this response length...
Do you have another idea?

hreintke, 2017-06-26:
Should not to be that difficult to overload `AsyncResponseStream` with something like 
`AsyncResponseStream::AsyncResponseStream(const String& contentType, size_t bufferSize, size_t maxBuffer)` which then stores the first maxBuffer bytes in cbuf and anything after that in SPIFFS/flash.
Pages smaller than maxBuffer are served directly from memory, larger using flash.
SPIFFS does wearing on the used flash so would prevent always writing to the same location.

jej, 2017-06-26:
@hreintke Thanks for your reply.
It remembers me these times when we did automodified assembly code on 68000 processor or copper list  generated by blitter :)
With this ASync, I feel like I put a 1200 horse power engine in my small car, and the car became so heavy that it doesn't roll anymore!
Well, probably perfect in some situations (I did video streaming with async) but here that's not adapted to my need. Currently I am more evaluating something than fighting to reach something. So either I give up my idea concerning nice GUI served by esp8266 (I could lighten my form to just one JSON field to manage that config), either I use something simple to do that without hacks.
Maybe I was a bit dreaming. When I saw the specs of esp8266 compared to atmega's, I thought everything was possible at no cost :) I will have think again! I understand that it's not so easy to develop async on so little hardware.
Thx.

me-no-dev, 2017-06-26:
Issue is the lack of memory, so you can not buffer the whole response and then send it asynchronously. for example sendString(a+b+c) will try to create a string of size the sum of the parts. There is nothing preventing the Async server to send a 10MB string in this case, other than the fact that the ESP can not allocate that much. The same exact thing is valid for any other amount or type of data that needs sending.

There s only one solution that I know of that could make you all happy, while the server stays async (in most cases) and that is the option to detach a Request from the async scheme and fill it in the loop (like the regular web server), but that is easier said than done in a way that would insure that users will not break the server (hanging connections will make the server inactive).

Buffering to SPIFFS is a bad idea :)

jej, 2017-06-26:
Thanks for the details. I understand a bit now.
The worry is that my sketch currently does quite few things. Just displaying 2 pages and one sends POST data to save it in a JSON file. I don't use so many variables for my needs. Half of the RAM is busy at start (see above, `Les variables globales utilisent 44704 octets (54%) de mémoire dynamique`). It seems only 32kb are available. Is it possible that my 4kb html page eats all of this just to be http processed?... (the symptom is that the page is partially displayed, cut in the middle). Maybe my assertion about RAM usage is false, after all...

me-no-dev, 2017-06-26:
@jej it's not the server that cuts your page ;) it's the esp that can not allocate all of the needed bytes (read above)

jej, 2017-06-26:
Ok, so if it's a RAM problem I will not investigate too long... no much hope that it changes. I don't know deeply the esp anyway.
By the way, I am surprised because when I compile an empty sketch for arduino I have 1% space program and 0% dynamic data occupied. On esp 51% / 38%. Probably different memory management scheme, I need to investigate that (not the place here). It cuts almost by two the idea I had on the esp power improvement :)
Thanks for your help.

me21, 2017-07-19:
Guys, won't ```AsyncWebServerRequest::sendChunked``` do what you want? It's just that, a callback function which returns the next chunk to be sent so you don't have to keep the entire response in RAM.

Kaisha, 2017-09-13:
I have to completely agree here with romanseidl, there really needs to be some way to 'push' data to a client, neither the standard sever nor Aysnc seem to be capable of it.  Buffering just doesn't make sense when you have so little ram to work with.

Continuations are, 'in theory' possible.  I fact in the Arduino code they use <cont.h> which supports continuations.  Thing is there is zero documentation, and I really don't want to spend the time trying to reverse engineer and create my own OS simply to server up a webpage.  Espressif has their own RTOS and a number of libraries for ESP8266/Arduino  support cooperative multitasking (which is just co-routines with a little scheduler on top).

Considering this library is an asynchronous library, I just assumed you used some sort of co-routine/scheduler in the background.  But it sounds like you implemented the whole thing on top of interrupts?

While I appreciate the time spent to write something like this, and that its a free project :)  That said Roman has a point, without some sort of push system you're not going to be able to have a real web server.

me-no-dev, 2017-09-13:
@Kaisha what are you talking about? the server can send megabytes of data... the fact that it does not do it the way you want it, does not make it incapable. If you are expecting a real OS level of server on a device with 60K of RAM.... you might be looking at the wrong place all together ;) If you want to use cont.h, then use the regular web server that runs on the Arduino thread. There already is such server and I see no point into writing another one. And push system.... would you be so kind to explain what "push system" is?

Kaisha, 2017-09-13:
Sorry, not trying to be critical, we do appreciate all the work you've put in.

The system you've written can be described as a 'pull' system.  ESPAsync library requests (pulls) data from the 'user' when it needs more data.  A push system is opposite, the user sends (pushes) data to the library.  To dynamically generate content, you need a push system (well 'need' is a bit strong, but its FAR easier).

An optimal solution would use some sort of continuation/scheduler (like <cont.h>) under the hood.  The library would call the callback, the call back would then write data in chunks.  Each write would then send the data immediately, and put the callback 'to sleep' running other tasks while data is being sent.  When the server is done sending data, you would wake back up the user callback thread, resuming where it left off, and it would continue to 'push' data until it is done.  I'm pretty sure this is what romanseidl and the others in this topic are talking about.

I'm not asking you to write it, that's asking a lot.  Nor am I attempting to criticize what you've done so far.  That said: "If you are expecting a real OS level of server on a device with 60K of RAM", actually you can do a lot.  If you look at https://github.com/espressif/ESP8266_RTOS_SDK, its a full real-time OS with optional preemeption (personally for small devices like this I would go with cooperative).

me-no-dev, 2017-09-13:
what you describe as push is exactly what the regular web server does. You fill a buffer and when it reaches certain size, it's sent and you start again once ACK is received. There is nothing Async in that ;) There is no need to point at RTOS either as I am developing the Arduino for ESP32 which is on top of RTOS and I have even ported ESP8266 Arduino to RTOS (only for myself). On OS it's a totally different thing ;) And the point of this server is not to push, but rather pull when it has the option to send more data. That way you can handle other connections at the same time when you don't have the option to spin up other threads.

Kaisha, 2017-09-14:
The problem with using ESP8266WebServer is that it seems (or at least I seem) to be having a problem with large responses.  A few kB is fine, but once I get into the 10's or more of kBs it restarts the chip.  It also seems to be far slower than your library.  I'm guessing it has to do with running out of memory.  sendContent doesn't seem to immediately send the data but rather buffers it internally?  Or at least that's my guess.  It also seems strange that both libraries use dynamic memory in many places where it just isn't necessary.

It seems that both ESP8266WebServer and ESPAsync cannot support large/dynamic pages, even though the ESP8266 can.  Any suggestions on a library that would allow it?

me-no-dev, 2017-09-14:
"though the ESP8266 can" ??? can how? You sound really knowledgable. Why not propose changes (code)? Let's see what you can do :) project is open and free ;) What is large dynamic page? Where are you going to store that dynamic data? And surely the regular web server will crap on you... you handle one connection at a time ;) but it "pushes" like you want.
For "async" pushing to work you need threads. And a "sync" request in each thread that you can fill and wait inside. ESP8266 has 2 threads! Arduino is one (cont.h) and everything else (network, async, system) is the other one.

Kaisha, 2017-09-14:
"though the ESP8266 can" ??? can how? You sound really knowledgable. "
Either English is not your first language, or you are taking this the wrong way.  No one in this thread is trying to be antagonistic, or insinuating you don't know what you are doing.  I know I certainly appreciate the time and effort spent, even if in the end its not the solution I will use.

"Why not propose changes (code)? Let's see what you can do :) project is open and free ;)"
It certainly has crossed my mind that I may have to write my own web server from scratch, I was just hoping I didn't have to...

"What is large dynamic page?"
100's kB to the MB range I would consider 'large' for an ESP8266.  At least something far larger than 80 kB.

"Where are you going to store that dynamic data?"
Well that's the point of this entire thread right?  There isn't enough ram on the ESP8266 to buffer pages that large, so the idea is to not store it, but rather generate it dynamically and send it off ASAP.  HTML is rather verbose, so a small amount of raw data can easily lead to large webpages.  Adding graphs, charts, scripts, all the bells and whistles of a proper webpage, a few kB's doesn't cut it.  The C++ source code to generate a web page is far smaller than the final result (well more accurately, the compiled result of C++ source code...).  1-2kBs of code and a few kBs of data can trivially generate MBs of html.

"And surely the regular web server will crap on you... you handle one connection at a time ;) but it "pushes" like you want."
Well there-in lies the problem, and why most people moved from ESP8266WebServer to ESPAsync.

"For "async" pushing to work you need threads. And a "sync" request in each thread that you can fill and wait inside."
I agree.  I thought that under the hood ESPAsync used threads, apparently not and just uses callbacks (again not a criticism, just an observation)?  My thinking was that if ESPAsync used <cont.h> threads under the hood, that it would be easy to add continuation style callbacks.  But if it doesn't (and from what you've stated I'm guessing it does not), then that would of course be difficult and probably an overly complex addition.

"ESP8266 has 2 threads! Arduino is one (cont.h) and everything else (network, async, system) is the other one."
Ya... so you do have knowledge of cont.h.  Is there any documentation around on it?  I can find nothing but a few uses deep in the Arduino source.  Its not even mentioned in the espressif documentation (at least that I can find).

Also if you know how to get the ESP8266WebServer to push more than a few kBs, I'd greatly appreciate it.  I've been trying using server.setContentLength(CONTENT_LENGTH_UNKNOWN); along with server.sendContent() but as soon as I get more than 30 or so kB it just crashes/restarts.

me-no-dev, 2017-09-14:
There is no documentation for cont.h and all of assembly inside is a bit above my level of understanding. It basically switches both "tasks". When you send large data through the regular server, you are blocking the network for too long and you hit issues, because packets are queueing for handling, but nobody is handling them.

me21, 2017-09-14:
@romanseidl @jej I'd like to remind you again of chunked responses. Plus protothreads (http://dunkels.com/adam/pt/) for coroutines. Try to give it a shot.

Kaisha, 2017-09-14:
"When you send large data through the regular server, you are blocking the network for too long and you hit issues, because packets are queueing for handling, but nobody is handling them."
Makes sense, well thank-you for the responses/insight.

romanseidl, 2017-09-14:
@me21 Unfortunately I don't have time to dig into this for the moment. But if someone would try to implement an example on how to combinine async-response and protothreads this would be really interresting.

Kaisha, 2017-09-16:
I got a solution.  Unfortunately I don't have the time to implement it ATM, but I'll throw it up here in case in the future someone stumbles across this and wants to implement it.

The template feature of ESPAsyn allows very simple dynamic webpages with a 'pull' style of programming.  If it were extended ever so slightly to allow for recursive templates you can have pretty much all the complexity of a push system, with a pull system.

What do I mean... well right now when ESPAsyn encounters a template in the form %ID% it calls a user supplied function to supply a string to replace %ID% with.  If that string was further recursed for templates (using the exact same call back) it would be pretty easy to create large dynamic content.

Implementation wise, a simple stack would handle saving the 'state' between chunked send/ack, the user of ESPAsync wouldn't need to manually create or manage the stack in anyway, it could entirely be handled by the Response class.  A new String type should also be used that supports dynamic string, string literals, and progmem literals in the same 'wrapper'.  Dynamically allocating large strings should be avoided (for obvious reasons) whenever possible.  Or some sort of type-tag system, some way to specify from the processor callback function that text is dynamic (needs free'ing), static (no free required) or progmem (no free required, aligned reads required), since they would need to be intermixed in the same function.

The function could look something like:
NewString processor(const NewString& var) {
   if (var=="HEAD") return "<head>%HEAD_TITLE%</head>";
   if (var=="BODY") return "<body>%NAV_BAR%%CONTENT1%%CONTENT2%%TABLE1%</body>";

   if (var=="HEAD_TITLE") return "...";

   if (var=="NAV_BAR") return "...";
   if (var=="CONTENT1") return "...";
   // etc...

   return "";
   }

Straightforward implementation, highly performant (the if's string matching could be replaced by a hash-table or a trie, but that's something that doesn't have to be supplied by ESPAsync and something the user of the library would be responsible for), no need for coroutines or threads, supports a pull-style data model, and simple to create large and highly dynamic html pages.

me21, 2017-09-16:
@Kaisha the part about recursive templates I do understand, but what do you mean by stack saving the state between send and ACK? What is it for?

Also, doesn't String already support progmem literals? Did you mean NewString should store only pointer to data, not copying the data to internal buffer, thus saving memory?

Kaisha, 2017-09-16:
"but what do you mean by stack saving the state between send and ACK? What is it for?"
As the templates are expanded, data will be written to some internal buffer till its full, and sent to the client that made the request.  At this point ESPAsync has to pause and wait for an ACK from the client before sending more data (there may be more than 1 packet in flight, but at some point ESPAsync has to pause and wait before continuing to send data).  In order to pause, it has to save its current state, which is the string its expanding.  In a recursive system, each string could expand to more strings, so the state has to be not only the current string, but previous strings and the position within them as well, so you would need a stack.  It would be simple (just string pointer, maybe a 'free string' destructor function pointer, and the current offset into the string) and could be rather shallow (even 10 entries deep would be a lot) and could be statically allocated, its size specified via a #define or template parameter (or even dynamically allocated); but the details aside, you would need some sort of stack to save the state in a recursive system.

"Also, doesn't String already support progmem literals? Did you mean NewString should store only pointer to data, not copying the data to internal buffer, thus saving memory?"
Exactly.  The whole point of dynamically generating HTML (as opposed to just buffering then sending it all at once) is to server pages that would otherwise be far too large for the ESP8266 memory to handle.  Large string will most likely be literals (both PROGMEM and normal literals) and smaller ones will probably be dynamic (either malloc/new char, or String class).  It would be necessary that the 'expansion' function callback return a type that can wrap any number of string types.  A simple templated type erasure wrapper would suffice, it doesn't need all the bells and whistles of a String class, just a way to access the data, and delete it if necessary.  I can throw one together quickly if needed (either as an example or to be used in ESPAsync desired).

The only reason I'm hesitant at coding the whole thing up is that at this point I just don't really know all the details of ESPAsync under the hood.  There's just so little documentation, and the source code tells you 'what' but never 'why', which makes it difficult to try and extent someone else's work.

terryjmyers, 2017-09-23:
@me-no-dev 
I have a large webpage loaded into a String (sometimes >6000bytes).  I've tried
`	AsyncResponseStream *response = request->beginResponseStream("text/html");
	response->print(w);
	request->send(response);
`
(where w is the large String)
This seems to work half of the time, but the other half I only get partial webpages, like the brower only loads the first half or second half.  Am I doing something wrong here?

Perhaps I need to use a beginChunkedResponse, but I'm having trouble following the example in the documentation as I don't have access to my String (w)  inside the function.  Could you provide a hint as to what I'm doing wrong

Here is my full callback function that serves a webpage
`void HTTPConfigGeneral(AsyncWebServerRequest *request) {

	String w = "";  w.reserve(7000);
	String msg = ""; msg.reserve(200);
	if (!is_authentified(request)) {
		AsyncWebServerResponse *response = request->beginResponse(301);
		response->addHeader(F("Location"), F("/login"));
		response->addHeader(F("Cache-Control"), F("no-cache"));
		request->send(response);
		return;
	}

	if (request->hasArg(F("USERNAME")) && request->hasArg(F("PASSWORD")) && request->hasArg(F("PASSWORD2"))) {
		if (request->arg(F("PASSWORD")) == request->arg(F("PASSWORD2"))) {
			EEPROMWriteHTMLLogin(request->arg(F("USERNAME")));
			EEPROMWriteHTMLPassword(request->arg(F("PASSWORD")));
			msg = F("Website login/password updated");
		}
		else {
			msg = F("Passwords don't match");
		}
	}

	if (request->hasArg(F("FontSize"))) {
		AsyncWebServerResponse *response = request->beginResponse(301);
		String cookie = F("FontSize="); cookie += request->arg(F("FontSize"));
		response->addHeader("Set-Cookie", cookie);
		response->addHeader("Location", "/Config");
		response->addHeader("Cache-Control", "no-cache");
		request->send(response);

		return;
	}

	if (request->hasArg(F("ProjectName"))) {//Process  a POST request to this webpage to change the project name
		if (!hasInvalidChar(request->arg(F("ProjectName")))) {
			Serial.println(F("Saving project name"));
			ProjectName = request->arg(F("ProjectName"));
			saveConfig();
			Restart();
		}
		else {
			msg = F("ERROR: Invalid Characters in Project Name.  Only alpha Numberic Characters");
		}
	}

	w += HTMLTop();
	w += HTMLMeta();
	w += HTMLTitle(F("Config"));
	HTMLHeader(request,w, false);
	w += HTMLContentBoxTop();
	w += F("		CONFIG>General<br><div id='comm' style=' background:black; height:3em; width:100% padding:5px; margin:5px;border:1px solid; vertical-align:center;' >PLEASE WAIT WHILE COMMUNICATION IS ESTABLISHED...</div>\n");
	w += HTMLContentBoxBottom();
	if (msg !="") { //Insert a row at the top for POST responses
		w += HTMLContentBoxTop();
		w += F("<h2 style='color:#F5B7B1;'>"); w += msg; w += F("</h2>");
		w += HTMLContentBoxBottom();
	}
[... 51 lines cut ...]
	w += HTMLContentBoxTop();
	w += F("\
		Change Project Name:<p>\n\
		<form action='/Config' method='POST'> \n\
			<input type='text' name='ProjectName' value='"); w += ProjectName; w+= F("' maxlength='32' required'><p>\n\
			<input type='submit' name='SUBMIT' value='Update Project Name'><p>\n\
			<p style='font-size:1em'>Note: Do not use any spaces, symbols, or puncuation marks, hyphens are fine.  Device will automatically reboot</p>\n\
		</form>\n");
	w += HTMLContentBoxBottom();
	w += HTMLContentBoxTop();
	w += F("font size: \
		    <a style='font-size:1em;' href='/Config?FontSize=1'>small</a>\n");
	w += F("<a style='font-size:1.5em;' href='/Config?FontSize=2'>medium</a>\n");
	w += F("<a style='font-size:2em;' href='/Config?FontSize=3'>large</a>\n");
	w += HTMLContentBoxBottom();
	
	w += F("		<script defer src='/jquery-3.2.1.min.js'></script>\n");
	w += F("		<script defer src='/Config.js'></script>\n");
	w += HTMLBottom();
	

	AsyncResponseStream *response = request->beginResponseStream("text/html");
	response->print(w);
	request->send(response);
`

me-no-dev, 2017-09-25:
Your String is a local variable and Async runs on top of callbacks, so in order Async to serve your string  like that, it would need to copy it in memory and keep it there together with the request/response until sent. One option is to "new" that string so it is in memory, attach it to the response using the public [_tempObject](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L188), use chinked to cut and send parts of it and free it at the end.

vanniaz, 2018-08-07:
**@me-no-dev** 
Hi, this is an old issue but I am writing here because I need a little help. As you suggested I am using **beginChunkedResponse** to serve a large JSON response which is generated on the fly from binary data on flash, and I am passing a specific function (genJsonLog) as callback:
`		AsyncWebServerResponse *response = request->beginChunkedResponse("text/json", [](uint8_t *buffer, size_t maxLen, size_t index) -> size_t
				{
					return genJsonLog((char *)buffer, (int)maxLen);
				}
			);
`
The problem is that the function must keep track of the current record in the flash file, which must persist from one call to the next. Now I declared it as a static variable and it works, but of course this is a bad solution since a second request, arriving right after the first one, would rewrite the variable.
You suggested to use **_tempObject** in Request, but I am not sure how to pass it to genJsonLog, can you give a hint?
Thanks
Giovanni

vanniaz, 2018-08-08:
I have managed to do it, I don't know if it's the best way (I am a C programmer and still have to learn C++ well), however it works:

```
request->_tempObject = new uint32_t;
*(uint32_t *)request->_tempObject = firstRecord;

AsyncWebServerResponse *response = request->beginChunkedResponse("text/json",
   [request](uint8_t *buffer, size_t maxLen, size_t index) -> size_t
	{
		return genJsonLog((char *)buffer, (int)maxLen, request);
	}
);
```

Then in the callback I retrieve current record information this way:

```
int genJsonLog(char *buffer, int maxLen, AsyncWebServerRequest *req)
{
	uint32_t *currRecord = (uint32_t *)(req->_tempObject);

     // use and increment *currRecord
     // ...
     // once the last record has been reached:
		delete (int *)req->_tempObject;
		return 0;

```
I hope this is ok (suggestions are welcome)

[7 more replies]