me-no-dev/ESPAsyncWebServer #190: Document how to use handleUpload and handleBody depending on what client sends

probonopd, 2017-07-23:
Assume I have an API endpoint `/api/files/local` that can be called with or without FILE.

The documentation says:
```
//Check if POST (but not File) parameter exists
if(request->hasParam("download", true))
  AsyncWebParameter* p = request->getParam("download", true);

//Check if FILE was uploaded
if(request->hasParam("download", true, true))
  AsyncWebParameter* p = request->getParam("download", true, true);
```

But the documentation does not say how to handle these two cases.

Consider this code:

```
  server.on("/api/files/local", HTTP_POST, [](AsyncWebServerRequest * request) {
    if(!request->hasParam("file", true, true)){
      // Need handleBody?
      // To call handleBody instead of handleUpload, we need a filter since we can only hardwire one handler to server.on. How to do this?
    }
    request->send(200, "application/json", "{}");
  }, handleUpload);
```

How can I make the upload work with `handleUpload` or `handleBody` depending on what the client sends?

probonopd, 2017-07-23:
This works:

```
  server.on("/api/files/local", HTTP_POST, [](AsyncWebServerRequest * request) {
    request->send(200, "application/json", "{}");
  }, handleUpload);
```
This does NOT work:

```
  server.on("/api/files/local", HTTP_POST, [](AsyncWebServerRequest * request) {
    request->send(200, "application/json", "{}");
  }, handleBody);
```
Plus how could I add a `if` statement to call `handleUpload` or `handleBody` depending on the situation?

[Here](https://github.com/probonopd/WirelessPrinting/commit/63858da3fd1bc22b99a57ebf6194d77a93eade09) is the full sketch.

probonopd, 2017-07-23:
The corresponding client code that does the uploading can be seen [here](https://github.com/fieldOfView/OctoPrintPlugin/blob/fc9cd78905af84b60c7c5f0a7cf5c9470253017e/OctoPrintOutputDevice.py#L372-L390).

probonopd, 2017-07-23:
Closing because https://github.com/me-no-dev/ESPAsyncWebServer/issues/191 addresses my needs. Turns out I don't need `handleBody`.

closed 2017-07-23 by probonopd

MichalFoksa, 2020-03-01:
Hi there,

I would like to re-open "how to use `handleBody(..)`" question and particularly, how to pass content of body from `handleBody(..)` to `handleRequest(..)`? 

My case is POST request with `text/plain` content type - not a JSON.

My understanding is that `handleBody(..)` is to parse content and `handleRequest(..)` is to act on it, or am I wrong?

My plan was to store request body in params, but it is not possible, cause `AsyncWebServerRequest::_addParam(..)` is private method.

```
void handleBody(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) {
  if (len == total &&  request->contentType() == "text/plain" ) {
    // :( Error: 'void AsyncWebServerRequest::_addParam(AsyncWebParameter*)' is private!!!
    request->_addParam(new AsyncWebParameter("body", String((char*) data), false, false, total));
  }
}
```
Help PLS.

sparkplug23, 2020-09-04:
@MichalFoksa Hey, did you ever figure this out? I am stuck trying to do the same thing and the documentation is not helpful.

MichalFoksa, 2020-09-04:
Hi @sparkplug23 
When I receive a payload I handle it (payload processing and a response send) in `handleBody(..)` and in `handleRequest(..)` I assume it was handled already.

See here, search for `POST /sys/master`: https://github.com/MichalFoksa/esphome/blob/feature/homeyduino-protocol/esphome/components/homeyduino/homeyduino.cpp

It works OK.

sparkplug23, 2020-09-04:
@MichalFoksa 

Thank you so very much for replying with code! I have been wrecking my head for days now trying to get mine to work and debugging the library to see how it works. I haven't yet tried your method, I will now, but sincere thanks for getting back to me so quickly.

MichalFoksa, 2020-09-04:
@sparkplug23 

You are welcome - happy to help. It is a bit frustrating that probably you have to identify request properties in both methods.

Good luck!
Michal

sparkplug23, 2020-09-04:
@MichalFoksa 

Forgive me for asking more. I actually was very close to what you had. My only confusion, and perhaps its my lack of knowledge of the virtual functions, but how does "handleBody" link to the incoming data. I see you have it placed inside a class, I understand that, but just how placing it there without (as far as I can tell) linking it at any point is confusing me. 

eg. nothing like this? 

```cpp
pWebServer->on("/connect", HTTP_POST, [](AsyncWebServerRequest *request) {
       request->send(204);
  }, NULL, handleBody);
```
Assume from your message that I need to "identify request properties" means only one handleBody will exist for the entire server, and not multiple ones per request.

Many thanks,
Michael

Leo-C, 2021-02-06:
Hi All,
also for me is a bit difficult to undersand behaviour of library from docs.

I'm developing an IoT application, exposing some API endpoint via AsyncWebServer, and I've developed a thin layer comfortable for me, deriving `AsyncWebHandler` that - given an URI - routes all GET requests on `hnd_get(AsyncWebServerRequest *request)`, `hnd_post(AsyncWebServerRequest *request, const String& body)` or `hnd_upload()` (with same parameters than `handleUpload()` )

To make this, after handler `handleBody()` and `handleUpload()` are called (as indicated in [docs](https://github.com/me-no-dev/ESPAsyncWebServer#request-life-cycle) ), I have to understand what kind of request is in `handleRequest()` to route to specific hnd_* handler.

After some trials & errors, I've discovered that:

- in some case for POST request `handleBody()` and `handleRequest()` are called in sequence , but in some case only `handleRequest()` is called (reading [code](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L131-L176) it seems that happens if it's a _PlainPost_ request)
- when only `handleRequest()` is called for a POST request, a parameter `body` contains all body content (but is also parsed if request is a Web Form POST)
- when a file is uploaded, some different parameter exists (`file`, `uploadedfile`, depending on browsers ...), but I've resolved checking if at least a parameter with File attribute exists with `if ((request->params() > 0) && request->getParam(0)->isFile())`

Than, a snippet of code that can help to understand what kind of request is, follows:
```
void handleRequest(AsyncWebServerRequest *request)
{
	//here some optional authentication check
	
	if (request->method() == HTTP_POST)
	{
		if ((request->params() > 0) && request->getParam(0)->isFile())
			; //NOP because handleUpload was called (and result was sent there)
		else if (_body != NULL) //_body is a String variable accumulated in handleBody()
		{
			hnd_post(request, *_body);
			delete _body;
			_body = NULL;
		}
		else //'isPlainPost' = true, then body is in param[0]
		{
			if (request->hasParam("body", true))
			{
				String body = request->getParam("body", true)->value();
				hnd_post(request, body);
			}
			else //POST with no content
				hnd_post(request, "");
		}
		
		return;
	}
	
	if (request->method() == HTTP_GET)
	{
		hnd_get(request);
		return;
	}
	
	request->send(500); //cannot be executed!
}
```

Many thanks for all your suggestions, and I'll be happy if someone can complete with other details my interpretation how to use ESPAsyncWebServer library

Leonardo