me-no-dev/ESPAsyncWebServer #194: Double memory allocation while sending data

jeremypoulter, 2017-07-24:
While developing an [application](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x) with ESPAsyncWebServer we are issues with content not downloading in AP mode. I have done a reasonable amount of looking in to this issue and I think it is down to memory usage.

I think I have traced one large contributing factor down to the LWIP library [tcp_out.c L375:L378](https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/core/tcp_out.c#L375:L378). This forces tcp_write to copy the data into internal buffers if ```LWIP_NETIF_TX_SINGLE_PBUF``` is 1, which it is [here](https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/include/lwipopts.h#L1194).

In my testing this means ~6Kb is needed to send data, ~3Kb allocated in [WebResponses.cpp](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebResponses.cpp#L273) and the other by tcp_write when copying that data. The biggest problem here is that if the first allocation succeeds and then tcp_write fails the data is lost and file reading gets out of sync and the request completes on the server side with little or no data actually sent to the client.

rick-bp, 2017-10-11:
I have the same problem!
#245

rick-bp, 2017-10-11:
#245

closed 2019-10-05 by stale[bot]