me-no-dev/ESPAsyncWebServer #197: Server gzipped files to safari bug, solution suggested

asexauer, 2017-07-29:
`bool AsyncFSWebServer::handleFileRead(String path, AsyncWebServerRequest *request) `
checks if the path with path.gz exists and than calls
`AsyncWebServerResponse *response = request->beginResponse(*_fs, path, contentType);`
with this new filename. This leads to a safari bug not decompressing the content for display.
The reason is that safari expects a Content-Disposition Header inline with the filename without the .gz extension. Chrome and Firefox are tolerating the .gz extension. AysncWebServerResponse handle this well, but due to the handling in AsyncFSWebServer::handleFileRead is called with the extension .gz and writing it to the Content-Disposition Header. So if I remove the handling for the .gz in AsyncFSWebServer::handleFileRead all works well. So I suggest to remove this here and leave the the .gz handling to  AsyncFSWebServer::handleFileRead.

changed code, just removed the lines 549-551 and 554 and 555by commenting them out.

```
bool AsyncFSWebServer::handleFileRead(String path, AsyncWebServerRequest *request) {
    DEBUGLOG("handleFileRead: %s\r\n", path.c_str());
    if (CONNECTION_LED >= 0) {
        // CANNOT RUN DELAY() INSIDE CALLBACK
        flashLED(CONNECTION_LED, 1, 25); // Show activity on LED
    }
    if (path.endsWith("/"))
        path += "index.htm";
    String contentType = getContentType(path, request);
    String pathWithGz = path + ".gz";
    if (_fs->exists(pathWithGz) || _fs->exists(path)) {
        //if (_fs->exists(pathWithGz)) {
        //    path += ".gz";
        //}
        DEBUGLOG("Content type: %s\r\n", contentType.c_str());
        AsyncWebServerResponse *response = request->beginResponse(*_fs, path, contentType);
        //if (path.endsWith(".gz"))
        //    response->addHeader("Content-Encoding", "gzip");
        //File file = SPIFFS.open(path, "r");
        DEBUGLOG("File %s exist\r\n", path.c_str());
        request->send(response);

        return true;
    } else
        DEBUGLOG("Cannot find %s\n", path.c_str());
    return false;
}
```

closed 2017-07-29 by asexauer

mkeyno, 2017-08-03:
@asexauer have you test this fix  with other browser?are  all browsers can handle gz extension ?

asexauer, 2017-08-04:
@mkeyno, I did a mistake, the problem was related to FSBrowserNG which is using AsyncWebServer Library. I posted the fix now there: https://github.com/gmag11/FSBrowserNG/issues/21
I tested it sucessfully with chrome and safari, safari on mac OS X and iOS.