me-no-dev/ESPAsyncWebServer #199: Memory leak when using async web server

paraflou, 2017-08-04:
I run my code with the sync ESP Web Server and got LmacRxBlk:1. 
In my case i have to send many http post requests to esp.
So i decided to use async web server. 

In the first case there was no memory leak.

Now using async web server every time i make a post request to esp heap memory gets consumed more in every request.

```c++
#include "SPISlave.h"
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ESP8266HTTPClient.h>
#include <MD_CirQueue.h>

#define HUE_LIGHTS_URL   "http://192.168.100.175/api/Zh969xxhyarhHjtj0QgWl-EEYyEDl4rAng3P-9xn/lights/3/state"

// WiFi parameters
const char* ssid = "trek-app";
const char* password = "trekc0n$2015";
byte mac[6];

// Incoming command data from spi
uint8_t loop_data[50];

int http_code;

// Queue parameters
const uint8_t  QUEUE_SIZE = 255;

// Async Web server
AsyncWebServer server(80); 

// Http Clients
HTTPClient hue_http;

// Concstruct Queue
MD_CirQueue Q(QUEUE_SIZE, 20 * sizeof(uint8_t));

void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Q.begin();

  // Connect to WiFi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" ----- WiFi Connected ----- ");
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  Serial.print(mac[0], HEX);
  Serial.print(":");
  Serial.print(mac[1], HEX);
  Serial.print(":");
  Serial.print(mac[2], HEX);
  Serial.print(":");
  Serial.print(mac[3], HEX);
  Serial.print(":");
  Serial.print(mac[4], HEX);
  Serial.print(":");
  Serial.println(mac[5], HEX);

 
  server.onRequestBody([](AsyncWebServerRequest * request, uint8_t *data, size_t len, size_t index, size_t total) {

    if (request->url() == "/spi") {
      Q.push(data);
      request->send(200, "text/plain", "end");
    }
  });

  server.begin();

  printf("SETUP heap size: %u\n", ESP.getFreeHeap());
}

void loop()
{
  if ((WiFi.status() == WL_CONNECTED) && !Q.isEmpty()) {

  Q.pop(loop_data);
    
  printf("--------HUE 1--------- Heap size: %u\n", ESP.getFreeHeap());
  hue_http.begin(HUE_LIGHTS_URL);
  http_code = hue_http.PUT((char *)loop_data);

  if (http_code == HTTP_CODE_OK) {
    Serial.printf("Http request ok\n");
  }

  hue_http.end();

  printf("--------HUE 2--------- Heap size: %u\n", ESP.getFreeHeap());
  }
}
```

What can i do?

me-no-dev, 2017-08-05:
Simple ;) you answer only when there is a body and the request is to /spi and in all other cases you do nothing, so you get lots of requests hanging. You need to answer to every request that you get

closed 2017-08-05 by me-no-dev

paraflou, 2017-08-05:
I never have any other case.

me-no-dev, 2017-08-05:
onRequestBody is not even the correct handler. this one only gives you the body of the request and can come in more than one packet. you need to define a normal on("/spi", HTTP_ANY, handler); to get the actual request once the bodyis parsed and request->send from there

paraflou, 2017-08-05:
Ok thanks a lot for the help! 
One last thing. How do I get the body of the request inside the handler that handles the normal on("/spi", HTTP_ANY, handler) ?

me-no-dev, 2017-08-06:
on("/spi", HTTP_ANY, handler, NULL, onBody)

paraflou, 2017-08-07:
Thanks me-no-dev but you are hard to follow. I saw ur source code and there is a function ::on that takes these arguments.
So do i have to call onBodyRequest somewhere? I assume that the above onBody parameter should be fed with an implementation of onBodyRequest.  Am i right?

paraflou, 2017-08-09:
I just tried it and it works! 
No memory leak any more.
Thanks a lot!

```c++
#include "SPISlave.h"
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ESP8266HTTPClient.h>
#include <MD_CirQueue.h>

#define HUE_LIGHTS_URL   "http://192.168.100.175/api/[key]/lights/3/state"

// WiFi parameters
const char* ssid = "trek-app";
const char* password = "trekc0n$2015";
byte mac[6];

// Incoming command data from spi
uint8_t loop_data[50];

int http_code;

// Queue parameters
const uint8_t  QUEUE_SIZE = 255;

// Async Web server
AsyncWebServer server(80); 

// Http Clients
HTTPClient hue_http;

// Concstruct Queue
MD_CirQueue Q(QUEUE_SIZE, 20 * sizeof(uint8_t));

void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Q.begin();

  // Connect to WiFi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" ----- WiFi Connected ----- ");
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  Serial.print(mac[0], HEX);
  Serial.print(":");
  Serial.print(mac[1], HEX);
  Serial.print(":");
  Serial.print(mac[2], HEX);
  Serial.print(":");
  Serial.print(mac[3], HEX);
  Serial.print(":");
  Serial.print(mac[4], HEX);
  Serial.print(":");
  Serial.println(mac[5], HEX);

  server.on("/spi", HTTP_ANY, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", "reqest end");
  }, NULL, [](AsyncWebServerRequest * request, uint8_t *data, size_t len, size_t index, size_t total){
                   // Do whatever you want with incoming data
   });

  server.begin();

  printf("SETUP heap size: %u\n", ESP.getFreeHeap());
}

void loop()
{
  if ((WiFi.status() == WL_CONNECTED) && !Q.isEmpty()) {

  Q.pop(loop_data);
    
  printf("--------HUE 1--------- Heap size: %u\n", ESP.getFreeHeap());
  hue_http.begin(HUE_LIGHTS_URL);
  http_code = hue_http.PUT((char *)loop_data);

  if (http_code == HTTP_CODE_OK) {
    Serial.printf("Http request ok\n");
  }

  hue_http.end();

  printf("--------HUE 2--------- Heap size: %u\n", ESP.getFreeHeap());
  }
}
```