me-no-dev/ESPAsyncWebServer #200: how to managing low heap condition within async request ? (or returning Error 5xx)

etiquet, 2017-08-06:
Dear All,

I have spent some time to try to understand my issue (ESP crashing) , seeking here to help or guidance. Apologies if this has already been solved or I missed something in the documentation but could not find something usefull.

In a ideal world the ESP (async webserver) should not not handle request if not enough avail ram and provide a 5xx error for example.

I tried to do it with :
```
 if ( ESP.getFreeHeap() < HEAP_SAFELIMIT ) 
        request->send(500, "text/plain","heap too low ");
```

But it seems, doing it while managing the request does not really helps.

_**More details below :**_

Based on the provided examples, I wrote a server that just provide back or set some internal variables in json or http get.
The code seems to work fine if request frequency is slow(seconds), but as soon the frequency increase the ESP8266 hang / reset.

Could go for about http 2 clients:
    - client-One is 10Hz getting the heap. ( http://192.168.1.114/heap )
    - client-Two is 100Hz requests/sec. ( http://192.168.1.114/getvariables/ )

In those conditions the free heap is about 27K ( indicated by client-one)

If a third requests comes in (http://192.168.1.114/setval?var=mastercontrol&val=735 ), after some transaction the ESP crash. With what seems a low heap condition. 
(created a vu meter to see quickly heap changes )

[image]

I tried to add some code to prevent lowheap condition but this not enough to prevent crash.
( multiple request in // could request memory async ? so not detecting it ?)

The code...
```
server.on("/heap",HTTP_GET, [](AsyncWebServerRequest *request){
        AsyncResponseStream *response = request->beginResponseStream("text/plain");
        response->addHeader("Access-Control-Allow-Origin", "*"); // to allow usage from local web page for test & development
        response->printf( String(ESP.getFreeHeap()).c_str() );
        request->send(response);
        });

server.on("/getvariables",HTTP_GET, [](AsyncWebServerRequest *request){
      if ( ESP.getFreeHeap() < HEAP_SAFELIMIT ) 
        request->send(500, "text/plain","heap too low ");
      
    AsyncResponseStream *response = request->beginResponseStream("text/json");
    response->addHeader("Access-Control-Allow-Origin", "*");
    DynamicJsonBuffer jsonBuffer;
    JsonObject &root = jsonBuffer.createObject();
    root["heap"] = ESP.getFreeHeap();
    int i;
    JsonArray& data = root.createNestedArray("pwmLamp");
    for (i=0; i<16;i++)
        data.add(pwmLamp[i]);
    root.printTo(*response);
    request->send(response);
    });

server.on("/setval",HTTP_GET, [](AsyncWebServerRequest *request){
      if ( ESP.getFreeHeap() < HEAP_SAFELIMIT ) 
        request->send(500, "text/plain","heap too low ");

      AsyncWebParameter* var = request->getParam("var");
      AsyncWebParameter* val = request->getParam("val");
      
    if(var){
        AsyncResponseStream *response = request->beginResponseStream("text/plain");
        response->addHeader("Server","ESP Async Web Server");
        response->addHeader("Access-Control-Allow-Origin", "*");
        if (!manageSetRequest(var,val)){
          response->printf("error in parram");
        } else 
          response->printf("Value setted");
        request->send(response); 
      } else 
        request->send(204, "text/plain","request not recognized");
    });

```

The stack says...

0x40214812: LinkedListNode at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebResponses.cpp line 471
:  (inlined by) LinkedList ::add(AsyncWebHeader* const&) at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/StringArray.h line 67
:  (inlined by) AsyncWebServerResponse::addHeader(String const&, String const&) at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebResponses.cpp line 108
0x402147e8: AsyncWebServerResponse::addHeader(String const&, String const&) at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebResponses.cpp line 471
0x40214812: LinkedListNode at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebResponses.cpp line 471
:  (inlined by) LinkedList ::add(AsyncWebHeader* const&) at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/StringArray.h line 67
:  (inlined by) AsyncWebServerResponse::addHeader(String const&, String const&) at /Users/etiquet/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebResponses.cpp line 108
0x402199bf: String::reserve(unsigned int) at /Users/etiquet/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/WString.cpp line 720
.../...

me21, 2017-08-31:
Try to apply https://github.com/esp8266/Arduino/pull/3533 locally - this should fix String low heap memory leak & crash issues.

me21, 2017-08-31:
Try to apply esp8266/Arduino#3533 locally - this should fix String low heap memory leak & crash issues.

philbowles, 2019-05-30:
This gets my vote to "bump" as I have similar (but worse 17k loss in 127mS) problem . We HAVE to have either a fix to this massive heap loos or the ability to set a threshhold below which it will ignore requests. My suggestions is:
The Retry-After response HTTP header indicates how long the user agent should wait before making a follow-up request. There are three main cases this header is used:

When sent with a 503 (Service Unavailable) response, this indicates how long the service is expected to be unavailable.
When sent with a 429 (Too Many Requests) response, this indicates how long to wait before making a new request.

...preferably the 429

philbowles, 2019-05-30:
You may be interested in this: https://github.com/me-no-dev/ESPAsyncWebServer/issues/521

closed 2019-10-05 by stale[bot]