me-no-dev/ESPAsyncWebServer #202: Delay response Json object

micha1983, 2017-08-11:
I Programmed a URL If this URL is called the Fibeln Parameter are Used to connect to a New wifi. 
One of this Parameters is If the New Connection should use DHCP or Not. 

As the response i would like to send a json with the ip Adresse given by the dhcp. 
But therefore i have to wait for the wifi Client Feedback. 
How can i Delay the response of the URL request until i Receive the ip of the dhcp. 

Delay is Not Working like written in the documents. 

Thanks
Michael

micha1983, 2018-07-14:
no not yet. i have#t followed this for a long time. but i'd would still be very nice. so if you will find something, let me please know.

cheers
Michael

rjlexx, 2018-08-05:
Please, check @hreintke answer regarding schedule_function here:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/383
May be it help you also.

closed 2019-10-05 by stale[bot]