me-no-dev/ESPAsyncWebServer #203: [Question] How to send a Async JSON Message to WebSocket Client

omersiar, 2017-08-14:
Hello  @me-no-dev , thank you for your great library. I'm an another noob to bother you,

https://github.com/me-no-dev/ESPAsyncWebServer#arduinojson-advanced-response
You have an example of sending 40KB Json Response, but i am failing to send it to a WebSocket client.

Server gets command messages from the websocket client and acts to it accordingly. It was working before where I dynamically create JSON buffer and sending it back to client, but now I have much more data (below 40KB) to send. I also tried to create a JSON object which has maximum 15 entries, then calling function one by one, this will require handling on client side which is not a problem. I trying to figure out best approach. If there is a way for example sending 2 MegaBytes data over WebSocket to client via (40KB) chunks i want to learn the way. 

```arduino
void sendPICClist() {
  AsyncJsonResponse * response = new AsyncJsonResponse();
  JsonObject& root = response->getRoot();
  root["command"] = "picclist";
  JsonArray& users = root.createNestedArray("list");
  Dir dir = SPIFFS.openDir("/P/");
  while (dir.next()) {
    JsonObject& item = users.createNestedObject();
    item["uid"] = dir.fileName();
    File f = SPIFFS.open(dir.fileName(), "r");
    size_t size = f.size();
    // Allocate a buffer to store contents of the file.
    std::unique_ptr<char[]> buf(new char[size]);
    // We don't use String here because ArduinoJson library requires the input
    // buffer to be mutable. If you don't use ArduinoJson, you may as well
    // use configFile.readString instead.
    f.readBytes(buf.get(), size);
    DynamicJsonBuffer jsonBuffer2;
    JsonObject& json = jsonBuffer2.parseObject(buf.get());
    if (json.success()) {
      String username = json["user"];
      int AccType = json["acctype"];
      unsigned long validuntil = json["validuntil"];
      item["username"] = username;
      item["acctype"] = AccType;
      item["validuntil"] = validuntil;
    }
  }
  ws.textAll(response, response->setLength());
}
```

`ws.textAll(response, response->setLength());`

me-no-dev, 2017-08-20:
uhmm... 2MB might be too much to do over WebSocket in one go ;) Using some synchronous communication in order to send the chunks and know that they are sent before sending again could do. Else you do not have nearly enough memory to do that on the ESP

omersiar, 2017-08-20:
```arduino
void sendUserList(int page, AsyncWebSocketClient * client) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["command"] = "userlist";
  root["page"] = page;
  JsonArray& users = root.createNestedArray("list");
  Dir dir = SPIFFS.openDir("/P/");
  int first = (page - 1) * 15;
  int last = page * 15;
  int i = 0;
  while (dir.next()) {
    if (i >= first && i < last) {
      JsonObject& item = users.createNestedObject();
      String uid = dir.fileName();
      uid.remove(0, 3);
      item["uid"] = uid;
      File f = SPIFFS.open(dir.fileName(), "r");
      size_t size = f.size();
      // Allocate a buffer to store contents of the file.
      std::unique_ptr<char[]> buf(new char[size]);
      // We don't use String here because ArduinoJson library requires the input
      // buffer to be mutable. If you don't use ArduinoJson, you may as well
      // use configFile.readString instead.
      f.readBytes(buf.get(), size);
      DynamicJsonBuffer jsonBuffer2;
      JsonObject& json = jsonBuffer2.parseObject(buf.get());
      if (json.success()) {
        String username = json["user"];
        int AccType = json["acctype"];
        unsigned long validuntil = json["validuntil"];
        item["username"] = username;
        item["acctype"] = AccType;
        item["validuntil"] = validuntil;
      }
    }
    i++;
  }
  float pages = i / 15.0;
  root["haspages"] = ceil(pages);
  size_t len = root.measureLength();
  AsyncWebSocketMessageBuffer * buffer = ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
  if (buffer) {
    root.printTo((char *)buffer->get(), len + 1);
    if (client) {
      client->text(buffer);
      client->text("{\"command\":\"result\",\"resultof\":\"userlist\",\"result\": true}");
    } else {
      ws.textAll("{\"command\":\"result\",\"resultof\":\"userlist\",\"result\": false}");
    }
  }
}
```

i guess i found a solution for this, simply sending the data chuck by chunk but each data is individual JSON object and has also an Array. Client handles the heavy lifting, i have tested with 1000 records, it works on Desktop with 15 records per paging, when i increase it to 30 records or more Mobile Chrome does not get the websocket message at all. I do not know what is failing, but desktop chrome gets it just fine.

https://github.com/omersiar/esp-rfid#tests

closed 2017-10-26 by omersiar