me-no-dev/ESPAsyncWebServer #204: Question about using spiffs and AsyncWebServer

davidk88, 2017-08-14:
In my sketch I'm reading some sensors and measurements are stored over multiple files on spiffs. Contents of these files are sent to browser as response to GET request:

`AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "filename.txt", "text/plain");
		request->send(response);`

The problem is that filesystem gets corrupted. Is this possible beacuse I'm writing to file from loop and in the same time web browser is reading that file. I read that spiffs is not thread-safe, but how can I synchronize access to filesystem?

Thank you very much

me-no-dev, 2017-08-20:
you can set a flag for example that would tell the web server that you are currently modifying the file and respond with some other code (I think there is html that you can send and that could cause the page to refresh after X seconds)

vanniaz, 2018-11-19:
I am experiencing exactly the same situation: a data logging application, where the same file is accessed both by the main loop (which writes the sampled data) and by the webserver (which reads the data and generates JSON content).  Occasionally the file system gets corrupted. I have put flags to avoid concurrent access to the same file, but the problem still appears from time to time.
Any suggestions? Is there a way to avoid AsyncWebServer to interrupt the main loop, just for a few milliseconds (the time required to write a new record in the file)? I could use noInterrupts() but I am not sure whether it is a good idea.
Sorry for posting to an old thread, but I see that it is still open so it is worth investigating.
@davidk88 have you found a workaround in the meantime?
Thanks

davidk88, 2018-12-06:
No, I went with synchronous webserver.

vanniaz, 2018-12-06:
@davidk88 thanks, in the meanwhile I found a couple of things:

- the corruption problems seem to have disappeared after I moved the web pages in the firmware image (which is a nice feature of EspAsyncWebServer). In other words, the webserver is not using SPIFFS for static web content anymore. The flash file system is only accessed by the "server.on" functions to read log data and output it in JSON. The same data is also accessed by the logging functions in the main loop, but concurrent access is avoided by means of synchronisation flags. This was not possible when the web pages were loaded from SPIFFS, because the webserver accessed them asynchronously and there was no way to synchronise access with flags.
- the above workaround would not have been necessary if concurrent SPIFFS access was handled correctly; in fact, SPIFFS has provisions for multi-threaded environments (with the SPIFFS_LOCK and SPIFFS_UNLOCK defines), I had not the time to check whether these defines are correctly used in the Arduino core and in EspAsyncWebServer, however it could be worth investigating, perhaps @me-no-dev can tell us more about this.

closed 2019-10-05 by stale[bot]