me-no-dev/ESPAsyncWebServer #206: Connect a simple websocket client from another ESP8266?

cotestatnt, 2017-08-14:
I have one ESP8266 working like webserver and i use it in order to sync GPIO status with connected clients (browser).
Now, i need to connect others ESP8266 acting like clients for exchanging data with server.
I saw there is a class named AsyncWebSocketClient, ho can i use it without adding other library?

me-no-dev, 2017-08-20:
It's not a client is the sense that you can use it to connect to a web socket server :) you need a websocket client lib for that. https://github.com/Links2004/arduinoWebSockets/tree/master/src

cotestatnt, 2017-08-20:
Ok, i'll do that.
Anyway, thank you for a so useful library!

closed 2019-10-05 by stale[bot]

ChangeD20, 2021-03-15:
> It's not a client is the sense that you can use it to connect to a web socket server :) you need a websocket client lib for that. https://github.com/Links2004/arduinoWebSockets/tree/master/src

I´ve tried to use the lib you mentioned with no succes.

@me-no-dev - could you please help out here?

https://github.com/me-no-dev/ESPAsyncWebServer/issues/956#issuecomment-799650192