me-no-dev/ESPAsyncWebServer #211: "Respond with content coming from a File containing templates" not working

UlliBien, 2017-08-30:
I tried the example for "**Respond with content coming from a File containing templates**". The callback method "**processor(..)**" is never called.

`request->send(SPIFFS, "/index.htm", String(), false, processor);` initializes a new intance of `AsyncAbstractResponse` class (`AsyncFileResponse` is derived from `AsyncAbstractResponse`). Field `_chunked` is set to true by the constructor. Via some other methods `AsyncAbstractResponse::_ack(..)` is called.

The problem is that `_ack(..)` does not call method `AsyncAbstractResponse::_fillBufferAndProcessTemplates(..) `which calls `_callback(..)` if `_chunked` is set to true. But it should! It calls `_fillBuffer(..)` instead. :-(

UlliBien, 2017-08-30:
This is a workarround for file "/t.svg":
```
...
  server.on("/t.svg", HTTP_ANY, handleCurrentTemperatureRequest);
...
```
```
void handleCurrentTemperatureRequest(AsyncWebServerRequest * request) {
  String sHTML;
  File PageFile = SPIFFS.open("/t.svg", "r");
  if (PageFile) {
    sHTML = readString(PageFile);
    PageFile.close();
    sHTML.replace("17,5", "20,1");
  }

  request->send(200,"image/svg+xml",sHTML);
}
```
With  `readString` (copied from Stream.cpp but with no yield):
```
// private method to read stream 
String readString(File s) {
  String ret;
  int c = s.read();
  while (c >= 0) {
    ret += (char) c;
    c = s.read();
  }
  return ret;
}
```

me21, 2017-08-31:
Oops, something got lost during pull request #189. Please check pull request #213, it should fix the bug.

commit e6c432e5 (2017-08-31): Something got lost during pull request #189. Fixes #211, updates documentation. (#213)

* Removed unnecessary memmove from chunked response generation.

* Added simple template processor to AsyncFileResponse.

Unzipped files in SPIFFS, Streams, PROGMEM strings, callback/chunked responses may have template placeholders like %TEMPLATE_VAR% inside. If callback is specified in Async...Response constructor call, it will be used to replace these with actual strings.
The prototype of callback is String(const String&), i.e. it gets variable name and returns its value.
Template variables' delimiter is  [... cut]
closed 2017-08-31 by me-no-dev