me-no-dev/ESPAsyncWebServer #212: How to share .bin file of ESP?

diffstorm, 2017-08-31:
Hi,

I want to share the firmware (.bin) file of ESP8266 using a web server as "application/binary" type. How can I do that using this library?

Thanks.

me21, 2017-08-31:
Well, I think you should write firmware file into SPIFFS and serve it as a static file. It's probably the easiest way.

diffstorm, 2017-08-31:
How can I write it to SPIFFS?

me21, 2017-08-31:
It depends on your IDE. Search its documentation. I'm not sure about Arduino IDE, but in Platformio you just create a subfolder named "data" in the project folder, then place needed files there and build SPIFFS image.

diffstorm, 2017-08-31:
The SPIFFS side is easy, how can I read the firmware into ram? Then I can write it as file. Which address? What size? etc..

me21, 2017-08-31:
You don't have to read file yourself to serve it. Read ESPAsyncWebServer docs to learn how to serve static files from SPIFFS, it's all there.

diffstorm, 2017-08-31:
@me21 You didn't understand me. My though is to read the firmware of ESP itself.
Also Arduino compiler says: Sketch uses 316,521 bytes (41%) of program storage space. Maximum is 761,840 bytes.
So this will not work with 256K SPIFFS.

me-no-dev, 2017-08-31:
you do not need to write the firmware to the file system ;) you can use chinked or similar method do read the flash and send it to the browser ;)

diffstorm, 2017-08-31:
@me-no-dev of course the original idea was what you said. Can you help with the addresses? I do not know where and how to read it byte per byte.

me21, 2017-08-31:
Hmm, this proved harder than I thought. Maybe https://github.com/espressif/esptool/wiki/Firmware-Image-Format will be useful.

diffstorm, 2017-08-31:
I cannot understand the whole thing in the link. First bytes are OK but I will copy them as they are, so I need to know where it starts and the size of binary. Also maybe a RAM function to read whole Flash without problems.

me21, 2017-08-31:
Perhaps https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map will be more useful?

diffstorm, 2017-09-01:
Both links are useful indeed, only I should understand them correctly. Which bin is the arduino sketch and how to read it etc.. I'll give a try some code to get it, I understand that nobody worked on this before. I have asked in other places too, but couldn't get exact answers. Thank you all.

hreintke, 2017-09-01:
Or in OTA. 
There they download a .bin file and copy the resulting image before restart

closed 2018-03-19 by diffstorm