me-no-dev/ESPAsyncWebServer #214: isssues

gusantor, 2017-09-01:
I'm having issues adding a header to an AsyncWebServerRequest

I'm using

`#include <ESP8266WiFi.h>
#include <ArduinoOTA.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "myssid";
const char* password = "mypass";
const char* hostName = "esp-async";

void setup(){
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  
  WiFi.hostname(hostName);
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP(hostName);
  
  WiFi.begin(ssid, password);
  
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("STA: Failed!\n");
    WiFi.disconnect(false);
    delay(1000);
    WiFi.begin(ssid, password);
  }

  //Send OTA events to the browser
  ArduinoOTA.onStart([]() { /*events.send("Update Start", "ota"); */});
  ArduinoOTA.onEnd([]() { /*events.send("Update End", "ota"); */});
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    char p[32];
    sprintf(p, "Progress: %u%%\n", (progress/(total/100)));
    //events.send(p, "ota");
  });
  ArduinoOTA.onError([](ota_error_t error) {
    /*if(error == OTA_AUTH_ERROR) events.send("Auth Failed", "ota");
    else if(error == OTA_BEGIN_ERROR) events.send("Begin Failed", "ota");
    else if(error == OTA_CONNECT_ERROR) events.send("Connect Failed", "ota");
    else if(error == OTA_RECEIVE_ERROR) events.send("Recieve Failed", "ota");
    else if(error == OTA_END_ERROR) events.send("End Failed", "ota");*/
  });
  ArduinoOTA.setHostname(hostName);
  ArduinoOTA.begin();

  server.on("/login", HTTP_GET, handleLogin);
  server.on("/login", HTTP_POST, handleLogin);

  server.begin();
}

void handleLogin(AsyncWebServerRequest *request){
  
  char titulo[100] = "Login";
  String contenido = "<div class=\"content\"><h1>Login</h1>";

  int headers = request->headers();
  int i;
  for(i=0;i<headers;i++){
    AsyncWebHeader* h = request->getHeader(i);
    //if(h->name().c_str()=="ESPSESSIONID"){
      Serial.printf("_HEADER[%s]: %s\n", h->name().c_str(), h->value().c_str());
    //}
  }

  int params = request->params();
  for(i=0;i<params;i++){
    AsyncWebParameter* p = request->getParam(i);
    if(p->isFile()){
      Serial.printf("_FILE[%s]: %s, size: %u\n", p->name().c_str(), p->value().c_str(), p->size());
    } else if(p->isPost()){
      Serial.printf("_POST[%s]: %s\n", p->name().c_str(), p->value().c_str());
    } else {
      Serial.printf("_GET[%s]: %s\n", p->name().c_str(), p->value().c_str());
    }
  }

  contenido += "<form action='/login' method='POST'>";
  contenido += "User:<input type='text' name='USERNAME' placeholder='user name'><br>";
  contenido += "Password:<input type='password' name='PASSWORD' placeholder='password'><br>";
  contenido += "<input type='submit' name='SUBMIT' value='Submit'></form>";
 
  contenido += "</div>";

  AsyncWebServerResponse *response = request->beginResponse(200, "text/html",  contenido);
  response->addHeader("Server","AB");

  request->send(response);
  
}

void loop(){
  ArduinoOTA.handle();
}`

but Serial just show this, and 'Server' header isn't in the output

`_HEADER[Host]: 192.168.1.157
_HEADER[Connection]: keep-alive
_HEADER[Content-Length]: 35
_HEADER[Cache-Control]: max-age=0
_HEADER[Accept]: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
_HEADER[Origin]: http://192.168.1.157
_HEADER[Upgrade-Insecure-Requests]: 1
_HEADER[User-Agent]: Mozilla/5.0 (X11; Linux i686) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36
_HEADER[Content-Type]: application/x-www-form-urlencoded
_HEADER[Referer]: http://192.168.1.157/login
_HEADER[Accept-Encoding]: gzip, deflate
_HEADER[Accept-Language]: es-419,es;q=0.8,en;q=0.6
_POST[USERNAME]: A
_POST[PASSWORD]: B
_POST[SUBMIT]: Submit
`

what I'm doing wrong ?

please, any help ?

thanks

me-no-dev, 2017-09-01:
In your code you respond with a header, so you should be searching for it in the browser :)

gusantor, 2017-09-01:
Hi @me-no-dev , thanks !, your answer helps me , but still have doubts about headers

I just though that my header ('Server') should appear in Serial monitor, among the others ('Host', 'Connection', 'Content-length', etc)

Following your answer, I review my browser and I realize that there are two groups of headers: 'Response' and 'Request', and my header ('Server') is there in the 'Response' Headers

but the origin of my test was to implement a simple authentication method ... as in SimpleAuthentification.ino (ho dude really thank you :) ), where you use somehting like this

    server.sendHeader("Set-Cookie","ESPSESSIONID=0");
    server.sendHeader("Set-Cookie","ESPSESSIONID=1");

is this feasible with asyncserver ?, how ?

thank you for all this work !!

me-no-dev, 2017-09-03:
it is totally possible the same exact way ;) difference is that "Set-Cookie" will set the "Cookie" header to that value and that "Cookie" will be present then next time that the browser connects to the esp

gusantor, 2017-09-06:
thank you thank you thank you :)

I'll give it a try on weekend

greetings !

gusantor, 2017-09-25:
Yep ! it works great ! thank you

closed 2019-10-05 by stale[bot]

eleminer, 2022-11-21:
Hi, i have the  same problem currently. How you send a header with "sendHeader"? @gusantor.
it says: 'class AsyncWebServer' has no member named 'sendHeader' :(

gusantor, 2022-11-21:
> Hi, i have the same problem currently. How you send a header with "sendHeader"? @gusantor. it says: 'class AsyncWebServer' has no member named 'sendHeader' :(

Sorry to say that whatever was the situation then, I don't have any clue how when I solved it.

And, as me-no-dev states, the Headers set with
response->addHeader where there, in the browser side, not in server (esp) side.

Please also note at ESPAsyncWebServer , Adding Default Headers

Greetings

eleminer, 2022-11-21:
@gusantor very sad, "addHeader" unfortunately does not work for me in the same way as sendHeader under esp webserver

wawanUnic, 2023-09-03:
Hi.
```
  server.on("/headerExamle", HTTP_GET, [](AsyncWebServerRequest *request){
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "This is example! Ok!");
    response->addHeader("Refresh", "3");
    request->send(response);
  });
 
```
src: [link](https://techtutorialsx.com/2018/07/24/esp32-arduino-web-server-add-header-to-response/).