me-no-dev/ESPAsyncWebServer #216: How to write to a file?

UlliBien, 2017-09-01:
I have a web site that sends some data via a parameter (...?data=TextToSave). Is there a method to write it to a SPIFFS file? Using `File::write(..)` calls yield and forces a reset of the ESP8266. Actually I save the data in a global variable and write it to the file inside `loop()`. I am concerned that this will lead to long-term synchronization problems.

me-no-dev, 2017-09-03:
interesting... we use File:write in the SPIFFS editor ;) works fine.
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L437-L453

UlliBien, 2017-09-04:
I'm very sorry. It was my fault. I called a method, that makes a file read with standard read method :-(

In the example i found this: `server.addHandler(new SPIFFSEditor(http_username,http_password));`
Bur how can I invoke the editor? I didn't find any hint. Neither on this GitHub project nor with google.

me-no-dev, 2017-09-04:
http://ESP_IP/edit

closed 2019-10-05 by stale[bot]