me-no-dev/ESPAsyncWebServer #217: File download inserts garbage

UlliBien, 2017-09-02:
Download for small files works well. But when the files are lager the first part of the file looks pretty good but for the rest garbage is inserted. The zip contains the original file (input.svg) and the downloaded file (output.svg). [files.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1272132/files.zip)

The code:

```
if (request->hasParam("filename", true)) { // Download oder Delete
  if (request->hasArg("download")) { // Datei download
    Serial.println("Download Filename: " + request->arg("filename"));
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, request->arg("filename"), String(), true);
    response->addHeader("Server", "ESP Async Web Server");
    request->send(response);
    }
```

me-no-dev, 2017-09-03:
The code you provide is not enough. It is really strange to get two responses entangled. What I can tell though is that I do not see if you respond at all if there is no param "download" and arg "filename"

UlliBien, 2017-09-04:
Hi,
here is a small project that demonstrates the behavior. If you open <IP>/config.html youget the following output: 
[image]
If you download a small file (e.g. "config.txt") it will be downloaded perfectly.
If you download a medium sized file (e.g. "t.svg") you will get a corrupted file.
if you download a lager file (e.g. "vp.svg") the download does not finish und the ESP8266 does not repond any more.

me-no-dev, 2017-09-04:
in ```handle_config``` you have a case where you return two responses ;) add return after the first ```request->send(response);```

me-no-dev, 2017-09-04:
```cpp
// ToDo: return notwendig? <<< YES! Return here
```

UlliBien, 2017-09-04:
Very much thx! That solved the problem. I looked over the code for several but didn't find the cuase.

closed 2019-10-05 by stale[bot]