me-no-dev/ESPAsyncWebServer #220: Spiffs OTA Update - Magic byte is wrong

ledsi, 2017-09-06:
Program upload works really fine with uploading the *.bin file and klick on update.
But if i select a spiffs.bin it starts uploading and after a while an error occours:

Update Start: spiffs.bin
ERROR[10]: Magic byte is wrong, not 0xE9

What am I doing wrong?
I used the code from example at readme.md

KR

ledsi, 2018-01-06:
Looks like there is a Solution for this...
https://github.com/esp8266/Arduino/commit/85973cdc8c2f6e00b35bd1f97648fa4b59ecf83e

Any way to add this changes on this project?

ledsi, 2018-11-25:
@me-no-dev 
any news here?

AcuarioCat, 2018-12-24:
There is a flag in the begin method that flags the update as either firmware or spiffs. You need to set it correctly.
Here is the code I use - only limitation is the filename has to have spiffs in it to work - but it gets named that when it is built..

`	if (!index) {
		String fn = filename;
		fn.toLowerCase();
		byte bSpiffs = fn.indexOf("spiffs") > 0 ? U_SPIFFS : U_FLASH;
		Serial.println("Update firmware");
		Update.runAsync(true);
		if (!Update.begin(free_space, bSpiffs)) {
			Update.printError(Serial);
		}
	}
`

closed 2019-10-05 by stale[bot]