me-no-dev/ESPAsyncWebServer #221: Template Engine Issue

ledsi, 2017-09-07:
Hi, tried to use the template engine but it isnt working.
Every time i request the page, the site loads the content, and the content and the content...

Finally i get the page with its content "test %replace% test" for hundret times (page didnt stop loading) and nothin will be replaced.

What I am doing wrong?

`
server.on("/test", HTTP_ANY, [](AsyncWebServerRequest *request){
	syncWebServerResponse *response = request->beginResponse(SPIFFS, "/test.htm", String(), false, processor);
	request->send(response);
});
String processor(const String& var)
{
  if(var == "HELLO_FROM_TEMPLATE")
	return F("Hello world!");
  return String();
}
`

me-no-dev, 2017-09-08:
Here is what I used and it worked:

template.htm
```html
<!DOCTYPE html>
<html>
  <head>
    <title>%HELLO_FROM_TEMPLATE%</title>
  </head>
  <body>
    <h2>%HELLO_FROM_TEMPLATE%</h2>
  </body>
</html>
```

code:
```cpp
String asyncProcessor(const String& var)
{
  if(var == "HELLO_FROM_TEMPLATE")
    return F("Hello world!");
  return String();
}

server.on("/template", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/template.htm", String(), false, asyncProcessor);
});
```

Result:
```html
<!DOCTYPE html>
<html>
  <head>
    <title>Hello world!</title>
  </head>
  <body>
    <h2>Hello world!</h2>
  </body>
</html>
```

ledsi, 2017-09-08:
Hm that did not work for me:

`%HELLO_FROM_TEMPLATE%
.1 200 OK Content-Type: text/html Content-Disposition: inline; filename="template.htm" Connection: close Accept-Ranges: none Transfer-Encoding: chunked aba
%HELLO_FROM_TEMPLATE%
.1 200 OK Content-Type: text/html Content-Disposition: inline; filename="template.htm" Connection: close Accept-Ranges: none Transfer-Encoding: chunked aba
%HELLO_FROM_TEMPLATE%
.1 200 OK Content-Type: text/html Content-Disposition: inline; filename="template.htm" Connection: close Accept-Ranges: none Transfer-Encoding: chunked aba
%HELLO_FROM_TEMPLATE%
.1 200 OK Content-Type: text/html Content-Disposition: inline; filename="template.htm" Connection: close Accept-Ranges: none Transfer-Encoding: chunked aba
%HELLO_FROM_TEMPLATE%
.1 200 OK Content-Type: text/html Content-Disposition: inline; filename="template.htm" Connection: close Accept-Ranges: none Transfer-Encoding: chunked aba
%HELLO_FROM_TEMPLATE%
.1 200 OK Content-Type: text/html Content-Disposition: inline; filename="template.htm" Connection: close Accept-Ranges: none Transfer-Encoding: chunked aba `

What is going wrong? I tried exact your example

me-no-dev, 2017-09-08:
maybe you are not running the latest code? Try updating the library ;) There was a fix recently that you might be missing

ledsi, 2017-09-08:
huh what a quick reply ;) thanks a lot.

Platform.io told me its the latest version, updated 21days ago.
If i click install again i get this message:

ESPAsyncTCP @1.0.1 is already installed

me-no-dev, 2017-09-08:
Maybe reinstall it? BTW you need to reinstall the AsyncWebServer ;) not AsyncTCP

ledsi, 2017-09-08:
Yes i did this, i always get this message when installing AsyncWebServer… :(

me-no-dev, 2017-09-08:
not using platformio really but I'm sure you can remove and reinstall the lib somehow. Or go inside it's folder and execute ```git pull```

ledsi, 2017-09-08:
oh yes, that worked for me, thanks.

However platform.io said its the newest version but it wasnt.
Now its fine... just a missing update *cry

Did a gread job, thanks 4 this ;)

closed 2017-09-09 by ledsi

DrP2016, 2017-10-08:
Hello,

i´m afraid i have to reopen this issue. Im running the latest versions of everything, but my file gets corrupted while being transferred. My .html file is about 20kb large and gets displayed correctly in all Browsers. As soon as i implement the templating mechanism, about 2kb of my .html file is missing and therefore the file can not be displayed properly. My placeholder %DEVICE_NAME% gets substitued correctly while beeing processed!

The code i use for processing is almost the same as the one in the example:
`String tplProcessor(const String& var){
  if(var == "DEVICE_NAME")
    return F("WebSocket Test!");
  return String();
}`
and
` server_async.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    DEBUG_PRINTLN("HTTP >> GET  >> '/'");
    request->send(SPIFFS, "/html/index.html" ); //, String(), false, tplProcessor);
  });`