me-no-dev/ESPAsyncWebServer #222: HTTP Persistent Connection and Pipelining

jeremypoulter, 2017-09-08:
Is [HTTP Persistent Connection](https://en.wikipedia.org/wiki/HTTP_persistent_connection) and [HTTP Pipelining](https://en.wikipedia.org/wiki/HTTP_pipelining) supported by ESPAsyncWebServer? If yes, is there an example? If no, where is the best place to start to implement it?

Stepping back a bit this is to help to loading a web page  using a .local (mDNS) host. Currently each resource is taking 2 seconds + to resolve the host name, so I am hoping using pipelining and/or persistent connections will drop down to a single host name resolution.

me-no-dev, 2017-09-08:
Maybe use WebSocket instead?

jeremypoulter, 2017-09-08:
This will probably work for some of the API calls, and indeed I have been starting to do this, but I am not sure how practical it is to download CSS, JS and images over WebSockets

me-no-dev, 2017-09-08:
I don't get why it takes 2s to resolve  .local host?

jeremypoulter, 2017-09-08:
I assume it is because these are using mDNS to resolve:

[image]

me-no-dev, 2017-09-08:
Maybe you can use NBNS on Windows so you can access it only by http://openevse/ :)
See here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266NetBIOS/examples/ESP_NBNST/ESP_NBNST.ino#L43

jeremypoulter, 2017-09-08:
> ESP8266NetBIOS/examples/ESP_NBNST/ESP_NBNST.ino#L43

Thanks, will give that a go.

Out of curiosity are you able to comment on the changes needed to support
at least persistent connections?

me-no-dev, 2017-09-10:
Not really sure TBH. I'm not positive that its a good idea on an mcu either :) you want to clear the memory ASAP and it's easy to have a connection running wild without knowing.

closed 2019-10-05 by stale[bot]