me-no-dev/ESPAsyncWebServer #223: platformio build on esp32 broken (library.json)

BlackEdder, 2017-09-09:
Because the library.json file includes a dependency on ESPAsyncTCP.h the library fails to build on esp32 hardware. This dependency should probably be hardware specific.

me-no-dev, 2017-09-10:
@ivankravets is this doable and how? I have [AsyncTCP](https://github.com/me-no-dev/AsyncTCP) that replaces ESPAsyncTCP on ESP32, so depending on the platform is which needs to be installed.

ivankravets, 2017-09-11:
I see 2 quick solutions:

1. Remove `dependencies` from manifest file
2. Create 2 different manifests for ESP8266 and ESP32 and specify custom dependencies. We can even keep the same name for library and just specify different "platforms" compatibility. 

What do you think?

me-no-dev, 2017-09-11:
I like the second solution, but how would the manifests be named so PIO picks them up?

ivankravets, 2017-09-11:
PIO does not depend strictly on `library.json`, it uses JSON content from a file.

I propose:

* `espressif8266_library.json`
* `espressif32_library.json`

Then, we will have 2 records in a registry. If a user specifies `lib_deps = ESPAsyncWebServer`, PIO will automatically install proper version depending on project settings.

Otherwise, we can have 2 different names to avoid "visual conflicts".

* `ESPAsyncWebServer`
* `AsyncWebServer`

BlackEdder, 2017-09-11:
In the documentation it says you can specify a platform for your dependencies[1]. Wouldn't that do the job, by adding both as a dependency with the relevant platform specified?

[1] http://docs.platformio.org/en/latest/librarymanager/config.html?highlight=dependencies#dependencies

me-no-dev, 2017-09-11:
@BlackEdder it does not support specifying the platform, so it would not work.
@ivankravets I will create two jsons :) give me a few minutes

ivankravets, 2017-09-11:
@BlackEdder  I'm laughing 😆 You are right! 

Yes, it seems that is will work. PIO will install both libraries, but only compatible will be used firstly.

@me-no-dev please merge https://github.com/me-no-dev/ESPAsyncTCP/pull/51

ivankravets, 2017-09-11:
@me-no-dev no no, @BlackEdder is right :)

me-no-dev, 2017-09-11:
@ivankravets i am confused now.... what should I do?

ivankravets, 2017-09-11:
@me-no-dev I can't make PR, you've removed the main `library.json` file. 

Please remove split files and use the one `library.json` with the next manifest:

```
{
  "name":"ESPAsyncWebServer",
  "description":"Asynchronous HTTP and WebSocket Server Library for ESP8266 and ESP32",
  "keywords":"http,async,websocket,webserver",
  "authors":
  {
    "name": "Hristo Gochkov",
    "maintainer": true
  },
  "repository":
  {
    "type": "git",
    "url": "https://github.com/me-no-dev/ESPAsyncWebServer.git"
  },
  "version": "1.1.0",
  "license": "LGPL-3.0",
  "frameworks": "arduino",
  "platforms":"espressif8266, espressif32",
  "dependencies": [
    {
      "name": "ESPAsyncTCP",
      "platforms": "espressif8266"
    },
    {
      "name": "AsyncTCP",
      "platforms": "espressif32"
    }    
  ]
}
```

BlackEdder, 2017-09-11:
I guess:
```
  "platforms":"espressif8266, espressif32",
```

should be:
```
"platforms":"espressif8266", "espressif32",
```

ivankravets, 2017-09-11:
@BlackEdder no no, it will be broken manifest. My version is valid.

me-no-dev, 2017-09-11:
done :) should be all good now! Thanks yall!

closed 2017-09-11 by me-no-dev

ivankravets, 2017-09-11:
@me-no-dev Please sorry that disturb you :( I've just checked all libraries and we have a small issue here. PIO Core works by default in [lib_compat_mode = 1](http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-compat-mode), which means, that we collect libraries and checking only for frameworks compatibility.

So, the current library will fail by default while a user doesn't set in `platformio.ini` `lib_compat_mode = 2`.

Sorry, please revert to this commit https://github.com/me-no-dev/ESPAsyncWebServer/commit/b7bf1fe9656fdc456c8d4d9aa10a0800ece38679 but change the description. Currently, it says that library is compatible for esp8266/esp32.  Need to specify in these manifests in the description that library is compatible only with X.

P.S: I recommend to use "espressif8266" instead of "espressif".

ivankravets, 2017-09-11:
@me-no-dev sorry, sorry, sorry again... I totally forgot what have created! :( 

- http://docs.platformio.org/en/latest/librarymanager/config.html#libldfmode

I'm going to test it now

ivankravets, 2017-09-11:
So, this comment is the final https://github.com/me-no-dev/ESPAsyncWebServer/issues/223#issuecomment-328586175

This is the best solution to keep compatibility with projects which use old PIO Core.

Please spit library into 2 manifests. Thanks!

me-no-dev, 2017-09-11:
Done :) split and corrected!

ivankravets, 2017-09-11:
@me-no-dev   Thanks!

@BlackEdder Please re-test with http://platformio.org/lib/show/1832/AsyncWebServer

jeremypoulter, 2017-09-15:
@me-no-dev  and @ivankravets I am now getting errors building for ESP8266, ESPAsyncTCP is no longer being installed (not a dependency of ESPAsyncWebServer). Is this expected and should I be adding an explicit dependency in my platformio.ini?

```
Library Dependency Graph
|-- <PubSubClient> v2.6
|-- <EEPROM> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266HTTPClient> v1.1
|   |-- <ESP8266WiFi> v1.0
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP Async WebServer> v1.1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <Hash> v1.0
|-- <DNSServer> v1.1.0
|   |-- <ESP8266WiFi> v1.0
|-- <Hash> v1.0
```

See https://travis-ci.org/jeremypoulter/ESP8266_WiFi_v2.x/jobs/274570571 for logs, etc

jeremypoulter, 2017-09-15:
As pointed out in https://github.com/platformio/platformio-core/issues/1050, I was using a direct reference to the Git repo (I was trying to ensure I had the bleeding edge), this change breaks that as there is now no longer any library.json. That is fine for me specific case but could be a problem if people want to reference forks rather than the official version.

ivankravets, 2017-09-15:
@jeremypoulter if you decide to manage libraries manually and use GIT versions, you can add `ESPAsyncTCP` as an extra dependency:

```
lib_deps =
  https://github.com/me-no-dev/ESPAsyncWebServer.git
  ESPAsyncTCP
```

jeremypoulter, 2017-09-15:
@ivankravets ok, cool I guess that will work, thanks

jeremypoulter, 2017-09-25:
@ivankravets FYI: I have just figured out why I was using the Git version. My platformio.ini has a specific version for most of the configs and I wanted to have one config that used the latest version for testing. However when using just ESPAsyncWebServer and ESPAsyncWebServer@be12e0c171 both libraries get checked out to .piolibdeps/ESPAsyncWebServer_ID306, when using the the git URL that gets checked out to .piolibdeps/ESPAsyncWebServer so there is no conflict.

I will change back to using the Git version and adding the extra dependency, but I presume there is an underlying bug/feature request against the Platform IO core that should come out of this?

ivankravets, 2017-09-25:
> both libraries get checked out to .piolibdeps/ESPAsyncWebServer_ID306, when using the the git URL that gets checked out to .piolibdeps/ESPAsyncWebServer so there is no conflict.

PIO Core does not rely on library folder name. It uses manifest inside it.

> but I presume there is an underlying bug/feature request against the Platform IO core that should come out of this?

Could you explain in details?

P.S: If you need to use custom libraries/versions per a few projects, you can use [extra library storage](http://docs.platformio.org/en/latest/projectconf/section_env_library.html#projectconf-lib-extra-dirs).

andig, 2017-10-01:
Checking in here. This doesn't work for me (https://github.com/platformio/platformio-core/issues/1098), even when using the git version of this repo.

andig, 2017-10-01:
@me-no-dev it's almost working. Could you add a

    {
      "name": "AsyncTCP",
      "platforms": "espressif32_stage"
    }    

to the dependencies? Otherwise it won't work with lib_compat_mode=2 and stage platform.

andig, 2017-10-01:
Or rather have

    "dependencies": [
      {
        "name": "ESPAsyncTCP",
        "platforms": "espressif8266"
      },
      {
        "name": "AsyncTCP",
        "platforms": ["espressif32", "espressif32_stage"]
      }    
    ]

as dependencies.

andig, 2017-10-01:
https://github.com/me-no-dev/AsyncTCP/issues/3 and https://github.com/me-no-dev/ESPAsyncWebServer/issues/239 should help with the staging version.

ivankravets, 2017-10-02:
I think we should move away from `espressif32_stage` name after PIO Core 3.5 release where we can depend on VCS (Git) URL directly and keep the same `espressif32` name of dev/platform.

For example,

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
```