me-no-dev/ESPAsyncWebServer #225: Using in other class (*.h/*.cpp)

vyruz1986, 2017-09-11:
I'm trying to use AsyncWebServer in a custom class I'm making.
My WebHandlerClass.h looks as follows:
```c++
// WebHandler.h

#ifndef _WEBHANDLER_h
#define _WEBHANDLER_h

#if defined(ARDUINO) && ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

#include <Hash.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

class WebHandlerClass
{
protected:
   AsyncWebServer _server;
   AsyncWebSocket _ws;
   void _WsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len);

public:
   void init(int webPort);
   void loop();
};

extern WebHandlerClass WebHandler;

#endif
```

And my WebHandler.cpp like this:
```c++
// WebHandler.cpp
#include "WebHandler.h"

void WebHandlerClass::_WsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t * data, size_t len)
{
   //removed for readability
}

void WebHandlerClass::init(int webPort)
{
   AsyncWebServer _server(webPort);
   AsyncWebSocket _ws("/ws");
   //ws.onEvent(std::bind(&Asy);
   //using namespace std::placeholders;
   _ws.onEvent(std::bind(&WebHandlerClass::_WsEvent, this
      , std::placeholders::_1
      , std::placeholders::_2
      , std::placeholders::_3
      , std::placeholders::_4
      , std::placeholders::_5
      , std::placeholders::_6));

   _server.begin();
}

void WebHandlerClass::loop()
{
}

WebHandlerClass WebHandler;
```

In my main *.ino file I have the following line in place to initialize the WebHandler class:  
`WebHandler.init(80);`

When I try to compile this I get the following error:  
`WebHandler.h: 17:7: error: no matching function for call to 'AsyncWebServer::AsyncWebServer()`

Which makes me think this library doesn't support setting the port number after initialization.

Is there a way I can initialize the `AsyncWebServer` object without port number and later on reinitialize it?

me-no-dev, 2017-09-12:
well there are two ways that you can go.
Either make the WebServer and WebSocket references and instantiate them in init, or instantiate them in the WebHandlerClass constructor, but then you can not init with different port

```cpp
class WebHandlerClass
{
protected:
   AsyncWebServer *_server;
   AsyncWebSocket *_ws;
   void _WsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len);

public:
   WebHandlerClass(){
     _server = NULL;
     _ws = NULL;
   }
   void init(int webPort);
   void loop();
};

void WebHandlerClass::init(int webPort)
{
   //check if already started maybe?
   _server = new AsyncWebServer(webPort);
   _ws = new AsyncWebSocket("/ws");
   //ws.onEvent(std::bind(&Asy);
   //using namespace std::placeholders;
   _ws->onEvent(std::bind(&WebHandlerClass::_WsEvent, this
      , std::placeholders::_1
      , std::placeholders::_2
      , std::placeholders::_3
      , std::placeholders::_4
      , std::placeholders::_5
      , std::placeholders::_6));

   _server->addHandler(_ws);
   _server->begin();
}

```

vyruz1986, 2017-09-13:
Thanks for the reply. 
Your suggestion about using pointers (At least I believe they're pointers... My c++- fu is not that strong :-( ) in the constructor makes the code compile at least.
Howerver when I try to visit the webserver (regular HTTP or websockets via the /ws path) I get a 500 internal server error.
Any idea's what could be causing this?
Everything works fine when I take the code out of the library and put it in the main *.ino file.

me-no-dev, 2017-09-13:
are you saying that the WebServer is running but the WebSockets are not? or both are not running?

vyruz1986, 2017-09-13:
The server is running but it returns a 500 internal server error.
So there is a connection being established but internally something is going wrong.

This happens both when I visit the regular webserver (http://<ip>) as well as the websockets page (http://<ip>/ws)

Is there any extra debug I can enable?

me-no-dev, 2017-09-14:
what if you add onNotFound handler to the server and see what comes out there? 500 usually means that it did not find a way to handle the request (no onRequest)

vyruz1986, 2017-09-17:
Sorry for the late reply.
I didn't have a lot of time over the past days, and found out that SPIFFS upload in ESP32 doesn't seem to work yet. I wanted to test first if a static index.html file would work instead of the websockets URL. I finally managed to upload an index.html file using the method described by bbx10 on another issue (https://github.com/espressif/arduino-esp32/issues/411#issuecomment-317908400).  

So I have a working index.html in my ESP32, this works fine and I can see the content when I browse to it.  
I added the onNotFound() handler as you requested, in the following way:
```c++
   _server->onNotFound([](AsyncWebServerRequest *request) {
      Serial.printf("Not found: %s!\r\n", request->url().c_str());
      request->send(404);
   });
```
This snippet was added in the `init()` function of the `WebHandler` class I created to which I've linked above.

As expected, when I try to browse to `http://<ip>/ws` the ESP32 logs on the serial line:  
`Not found: /ws!`

Aarghh.... As I was typing this post I realized where my problem was...

I forgot to add `_server->addHandler(_ws);` after setting the `_ws.onEvent()` handler. 

So problem fixed, thanks for you help and providing me the imaginary rubber duck to debug with :)

me-no-dev, 2017-09-18:
hahahahah!!! great! Closing this issue now

closed 2017-09-18 by me-no-dev