me-no-dev/ESPAsyncWebServer #226: Intermittent data loss

UlliBien, 2017-09-12:
I do need your help one more time. I've tried to reduce the code to a minimum to show the problem.

I have a webserver project [AsyncMinServer.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1295358/AsyncMinServer.zip) with 8 files in SPIFFS (see directory "data").
"index.html" is then main file. It has one template "%location%". There are references to a favicon and to six image files of type SVG. Each of this SVG files contains two templates "%wert%" and "%zeiger%".

For the replacement of the templates I wrote handler functions (in "handlers.cpp"). One for the index file and one for the SVG. The inserted values are dummies. E.g.:
```
String SvgProcessor(const String& var) {
  if (var == "zeiger")
    return "<use xlink:href=\"#Zeiger\" transform=\"rotate(45, 125, 125)\"/>";
  if (var == "wert")
    return String(99);
  return String();
}

void handle_Svg(AsyncWebServerRequest * request) {
  request->send(SPIFFS, request->url(), "image/svg+xml", false, SvgProcessor);
}
```

When loading the Web site via Firefox very often one of the images is missing (see "Firefox.png" in 
[images.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1295404/images.zip)). The same is true for Internet Explorer (see "IE.png"). I could not reproduce the problem with Chrome. But Chrome often shows five of these images an then pauses for 1 to 2 seconds and then the last image is displayed.

I increased the max open SPIFSS files to 10, but this has no effect.

I'm using Visual Micro (a Visual Studio plugin) for project compilation. Here is the full VM projekt: 
[AsyncMinServerVM.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1295430/AsyncMinServerVM.zip)

I have two hints:
1) I replaced the SVG files with a dummy image and loaded the files with java script with an interval of 100 ms between each of the files. This solves the problem, but it looks terrible when one image appears after the other. HTML code:
```
...
<div id="t" style="top:150px;left:1px"><img alt="" src="1pixel.png"/></div>
...>
<script>
    function foo (id){ document.getElementById(id).innerHTML 
                              =  '<img alt="" src="' + id + '.svg" />'; }
    window.setTimeout("foo('t')", 100);
...
</script>
```

2. I added a debug print to `size_t AsyncClient::add(const char* data, size_t size, uint8_t apiflags)` in ESPAsyncTCp.cpp to see what is transmitted:
```
...
  size_t will_send = (room < size) ? room : size;

  // -- Debug start
  Serial.println(">>>>>");
  Serial.write(data, will_send);
  Serial.println("<<<<<");
  // -- Debug end

  int8_t err = tcp_write(_pcb, data, will_send, apiflags);
...
```
Here is the log: [log.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1295530/log.zip)

It seems that the individual chunks of the file are not transferred one after the other. They are interlaced. Especially the last chunk with size 0 is often delayed. May be this causes the problem. In Firefox the busy symbol [image] occurs for a long period some times even if all images are shown correctly.

me-no-dev, 2017-09-12:
You are requesting 6 files at once and you have SPIFFS limit of 5. You hit the limit and some browsers can not handle it ;)
Just FYI, SVGs are Dom objects that you can manipulate with JS. So you can display the images and just change the data to show different values.

closed 2017-09-12 by me-no-dev

UlliBien, 2017-09-12:
Thanks for the fast answer. But may be you are wrong? I increased the SPIFFS limit to 10 (compiling with "SPIFFS_MAX_OPEN_FILES=10" at command line level, see above). SPIFFS.info maxOpenFiles gives 10.

For a test, I reduced the limit to 2 files. This increases the problem. A number of 15 does not solve it. After reading [https://github.com/me-no-dev/ESPAsyncWebServer/pull/193](url) I think that in case of exceding SPIFFS file limit, the system will call onNotFound handler. But there is no call to that function.

I already considered your advice. But in that case I cannot combine the files simply in another way. Anyway, if this is the only solution I will try it.

Another question: When replacing templates, what are the limits? In another file I want to insert a large table content. In the middle of the replacement a lot of chars are lost. Do you have a hint for me?

me-no-dev, 2017-09-12:
Templates are not my code, so I will probably need to go over it and see if I can make it work with larger strings. BTW if you do not hit the Max Open Files, you are hitting the connections limit ;) AFAIK it's also 5.

UlliBien, 2017-09-16:
Hi, it took me some time to reproduce the error in a small project: [ESP8266_W.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1308324/ESP8266_W.zip).
This is the HTML source:
[image]

... and this the received file (true for Firefox and Chrome): 
[image]

This is the complete defective page source saved from firefox: [ESP-Response.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/1308333/ESP-Response.txt)

For me, it seems that the error depends on the size of the file and as well as on the size of the replacements. My assumption is buffer overflow or wrong calculation of a pointer into a buffer .

I tried to localize the problem but there are to many things I don't have any experience with.

UlliBien, 2017-09-16:
I found the solution for this problem. It occurs in 
`size_t AsyncAbstractResponse::_fillBufferAndProcessTemplates(uint8_t* data, size_t len)` when there two or more replacements with length of replament value > length of replacement name. 

In case of value size larger then name size the tail of the already read buffer is shiftet to an internal cache. The next reading then first goes from cache and when cache is read then from file.

If there is another replacement with length of replament value > length of replacement name, another shift is nessecary. In the original text this additional shifted text was placed **before** the already shifted data. But the function **appends** it to the exiting cache. This swaps some portions of the data.

Tis is the code:
```
std::vector<uint8_t>::iterator i = _cache.end();
...
size_t pos = i - _cache.begin();
_cache.insert( i, //  <--- This is the problem
              &data[originalLen - (pTemplateStart + numBytesCopied - pTemplateEnd - 1)], 
              &data[len]);
i = _cache.begin() + pos;
...
```
Replacing the line containing `cache.insert` with
```
_cache.insert(_cache.begin(),
              &data[originalLen - (pTemplateStart + numBytesCopied - pTemplateEnd - 1)], 
              &data[len]);
```
solves the problem.

There are other references to ** i **. May be there are other unwanted side effects.

UlliBien, 2017-09-16:
The size of template name must not exceed 32 chars. 
`#define TEMPLATE_PARAM_NAME_LENGTH 32`
You should add this to the documentation

UlliBien, 2017-09-17:
It was hard to understand how `AsyncAbstractResponse::_fillBufferAndProcessTemplates` works: 
The internal `vector<uint8_t>_cache` together with the file stream work like a stack. `_readDataFromCacheOrContent` reads from this stack and `_cache.insert(...)` puts unused data back to the stack. So all inserts have to go to the top of the stack. Top of stack is the beginning of the vector. So all inserts should start with `_cache.insert(chache.begin(), ...)`.

I did this for all occurrences of `_cache.insert` and eliminated the iterator `i`. Seems to work as expected now.

This is the code: 
[_fillBufferAndProcessTemplates.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/1308952/_fillBufferAndProcessTemplates.txt)

me-no-dev, 2017-09-18:
why not offer this as PR? :)

UlliBien, 2017-09-18:
Sorry, but I'm not familiar with github. It already took me a lot of time to exlore how to make comments.

me-no-dev, 2017-09-19:
OK I have it and can make this into a branch to be tested :)