me-no-dev/ESPAsyncWebServer #227: One more time please me-no-dev

wcat13, 2017-09-13:
I am really sorry to bother you again

I thought I would be able to figure the second half after yesterdays bailout.  Alas I cannot.

can you(or anyone else) rewrite posted() for me.

```
String input(AsyncWebServerRequest *request, String argName) {
  String a=request->arg(argName);
  a.replace("<","&lt;");
  a.replace(">","&gt;");
  a.substring(0,200); 
  return a; 
}

String posted() {
  String msg=input("m"); 
  allMsgs="<li>"+msg+"</li>"+allMsgs;
   
  return header(POSTEDTITLE) + POSTEDBANNER + "<article>"+msg+"</article><a href=/>Back to index</a>";
}
```

Thank you & terribly sorry to bother you with this...

me-no-dev, 2017-09-13:
No, after what I gave you yesterday you should be able to do it. You need to pass along the request. what is ```header(POSTEDTITLE)```?
The way I rewrote your function yesterday is the way you need to rewrite those. The same exact thing. If that proves too much for you, I will advise you to steer away from Async ;) It is kinda really advanced and you will have more issues than success. If you still want to though ;) you have all necessary info how to do it (and the README)

closed 2017-09-13 by me-no-dev

wcat13, 2017-09-14:
Fair enough. As I would like to understand this I will continue to throw shit at it.  lol, Will assume you gave me some sort of hint in the form of a question.  Thank you