me-no-dev/ESPAsyncWebServer #228: CAPTIVE PORTAL ON ESPASYNCWEBSERVER (LMACRXBLK:1 ERROR)

treii28, 2017-09-13:
I'm trying to figure out how to set up a captive portal using the ESPAsyncWebServer and not really getting anywhere. The only information I can seem to find on the net says to start the dns server with an 'asterix' glob for the hostname pointing back to the device IP:

`dns.start ( 53, "*", WiFi.softAPIP() );`

But as soon as I add this to my code and try to connected to the soft_AP, the console starts spitting out `LmacRxBlk:1` over and over indefinitely and doesn't do anything else until I reset it. I have narrowed it down specifically to this dns line suggested by the other posts. I assume something is getting stuck in a loop, but I'm not sure what needs to be done differently as there is little or not other information I can seem to find.

I would really like to be able to trigger handheld devices such as android phones and iphones to auto-pop-up the web page of my devices as I'm trying to make a dumbed down device for random users to connect to to sign in at a specific location to say they were there. The less instructions I have to give, the better. Any help to getting a captive portal to work with the ESPAsyncWebServer code is greatly appreciated.

`// needed for platformio
#include <Arduino.h>
// needed for async webserver that doesn't seem to find Hash on platformio
#include <Hash.h>
#include <ESPAsyncWebServer.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>

#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>

const char* host = "FlashAsync";
const char* ssid = "ESPWiFi";
IPAddress ip ( 10, 10, 10, 1 ); // for softAP mode if used

AsyncWebServer server(80);
DNSServer dns;

// redirect used for any specific device requests for captive portal handling
void captivePortalTarget(AsyncWebServerRequest *request) {
    request->redirect("/index.htm");
}

// the following three functions are run in order from setup
void initWiFi() {
    WiFi.hostname(host);
    WiFi.mode(WIFI_AP);
    WiFi.softAPConfig ( ip, ip, IPAddress ( 255, 255, 255, 0 ) );
    WiFi.softAP(ssid);
    // if uncommented, the following line produces LmacRxBlk:1 errors
    //dns.start ( 53, "*", WiFi.softAPIP() ); // captive portal redirect everything to here when in AP mode
}

void initMDNS() {
    if (MDNS.begin(host)) {
        MDNS.addService("http", "tcp", 80);
        DBG_OUTPUT_PORT.println("MDNS responder started");
        DBG_OUTPUT_PORT.print("You can now connect to http://");
        DBG_OUTPUT_PORT.print(host);
        DBG_OUTPUT_PORT.println(".local");
    }
}
void initWebserver() {
    server.on("/delete", HTTP_DELETE, handleDelete);
    // upload a file to /upload
    server.on("/upload", HTTP_POST, returnOK, handleUpload);
    // Catch-All Handlers
    server.onFileUpload(handleUpload);
    //server.onRequestBody(onBody);

    //Android captive portal. Maybe not needed. Might be handled by notFound handler.
    server.on ( "/generate_204", captivePortalTarget );
    //Microsoft captive portal. Maybe not needed. Might be handled by notFound handler.
    server.on ( "/fwlink", captivePortalTarget );

    // my base server code uses a variation of the SD webserver to load content
    //  from spiffs/flash filesystem finding files when 'not found' is triggered if there are
    //  not specific handlers defined. An index.htm is included on the filesystem
    server.onNotFound(handleNotFound);

    server.begin();
    DBG_OUTPUT_PORT.println("HTTP server started");
}`

me-no-dev, 2017-09-13:
you need to call ```dns.processNextRequest();``` in the loop

closed 2017-09-13 by me-no-dev

wcat13, 2017-09-14:
This works nicely for me.  I never got around to figuring out how to get them off the built in browser for captive portals and back onto their own browser fluidly, but this will redirect any http request to index.html
```
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <FS.h>         
#include <ESPAsyncTCP.h>    
#include <ESPAsyncWebServer.h> 
#include <ESP8266mDNS.h>   

unsigned long previousMillis = 0;
int ledState = LOW;
int interval = 1000;
const byte DNS_PORT = 53;
DNSServer dnsServer;

IPAddress apIP(192, 168, 4, 1);
AsyncWebServer server(80);

void setup() {
  Serial.begin(115200);
  Serial.println();
  pinMode(LED_BUILTIN, OUTPUT);

  SPIFFS.begin();

  dnsServer.setErrorReplyCode(DNSReplyCode::NoError);
  dnsServer.start(DNS_PORT, "*", apIP);

  server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=600");

  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->redirect("/index.html");
    delay(100);
  });

  server.on("/generate_204", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->redirect("/index.html");
    delay(100);
  });

  server.on("/fwlink", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->redirect("/index.html");
    delay(100);
  });

  server.onNotFound([](AsyncWebServerRequest * request) {
    request->send( 200, "text/html", "URI Not Found" );
  });

  server.begin();
  WiFi.hostname("asynccaptive");
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("async captive","",10);  // open on channel 10

  if (!MDNS.begin("Project911Awareness")) {
    Serial.println(F("Error setting up MDNS responder!"));
  }
  // Add Web Server service to mDNS
  MDNS.addService("http", "tcp", 80);

}

void loop() {
  dnsServer.processNextRequest();
  unsigned long currentMillis = millis();
  int clients = WiFi.softAPgetStationNum();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    if (ledState == HIGH && clients >= 1) {
      ledState = LOW;
      Serial.printf("Stations connected = %d\n", clients);
    } else {
      ledState = HIGH;
    }
    digitalWrite(LED_BUILTIN, ledState);
  }
}
```

me-no-dev, 2017-09-14:
do I see delay in the Async callback? :D NO-NO-NO!!!

me-no-dev, 2017-09-14:
and @wcat13 you will be better off using [rewrite](https://github.com/me-no-dev/ESPAsyncWebServer#rewrite-to-different-index-on-ap) instead of redirect.

me-no-dev, 2017-09-14:
Here is a small captive test:

```cpp
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include "ESPAsyncWebServer.h"

DNSServer dnsServer;
AsyncWebServer server(80);

class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {}
  virtual ~CaptiveRequestHandler() {}

  bool canHandle(AsyncWebServerRequest *request){
    //request->addInterestingHeader("ANY");
    return true;
  }

  void handleRequest(AsyncWebServerRequest *request) {
    AsyncResponseStream *response = request->beginResponseStream("text/html");
    response->print("<!DOCTYPE html><html><head><title>Captive Portal</title></head><body>");
    response->print("<p>This is out captive portal front page.</p>");
    response->printf("<p>You were trying to reach: http://%s%s</p>", request->host().c_str(), request->url().c_str());
    response->printf("<p>Try opening <a href='http://%s'>this link</a> instead</p>", WiFi.softAPIP().toString().c_str());
    response->print("</body></html>");
    request->send(response);
  }
};

void setup(){
  //your other setup stuff...
  dnsServer.start(53, "*", WiFi.softAPIP());
  server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);//only when requested from AP
  //more handlers...
  server.begin();
}

void loop(){
  dnsServer.processNextRequest();
}
```

prstprst, 2019-04-15:
Hello, 
this work for me ... very simple : asynwebsever (for eps32) server.onnotfound used for captive portal 👍 

`/*********
  Rui Santos
  Complete project details at https://randomnerdtutorials.com  
*********/

// Import required libraries
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include "SPIFFS.h"
#include <DNSServer.h>

// Set LED GPIO
const int ledPin = 2;
// Stores LED state
String ledState;
const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 199, 1);

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);
DNSServer dnsServer;

// Replaces placeholder with LED state value
String processor(const String& var){
  Serial.println(var);
  if(var == "STATE"){
    if(digitalRead(ledPin)){
      ledState = "ON";
    }
    else{
      ledState = "OFF";
    }
    Serial.print(ledState);
    return ledState;
  }
  return String();
}
 
void setup(){
  // Serial port for debugging purposes
  Serial.begin(115200);
  pinMode(ledPin, OUTPUT);

  // Initialize SPIFFS
  if(!SPIFFS.begin(true)){
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }

  // Creating Wi-Fi
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

 
  dnsServer.start(DNS_PORT, "*", apIP);
  
  // Route for root / web page
  server.onNotFound( [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });
 
   // Route to load style.css file
  server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/style.css", "text/css");
  });

  // Route to set GPIO to HIGH
  server.on("/on", HTTP_GET, [](AsyncWebServerRequest *request){
    digitalWrite(ledPin, HIGH);    
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });
  
  // Route to set GPIO to LOW
  server.on("/off", HTTP_GET, [](AsyncWebServerRequest *request){
    digitalWrite(ledPin, LOW);    
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });
  
 
    // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request

// Start server
  server.begin();
}
 
void loop(){
  dnsServer.processNextRequest();
  
}`

samuk, 2019-06-18:
The solution from @prstprst compiles and runs, I can access the server, however, the captive portal does not redirect. Has something changed in the underlying libraries?

cnkz111, 2020-02-15:
The solution of @prstprst crashes in my setup (ESP32 V1.0.4, Arduino 1.8.12)
The solution of @me-no-dev  works fine.

I found the root cause for the crashes, it is the `WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));` call. It works fine without that call