me-no-dev/ESPAsyncWebServer #229: “Connection refused” when send GET-request from ESPAsyncWebServer

topersober, 2017-09-13:
Hello everyone! I have some TEST function that sends GET-request from one ESP to another.
(Both of them are working on ESPAsyncWebServer)
1) When I enter http://192.168.43.208/led=1 from Google Chrome – all fine 
2) When I call TEST function from Standart ESP8266WebServer - all fine
3) When I call TEST function before starting ESPAsyncWebServer (calling server.begin();) - all fine, second ESP responsing 200.
but...
4) When i call TEST function after staring  ESPAsyncWebServer, I get error code (-1)
( switch off AsyncWebSocket and AsyncEventSource - promlem remains.
SO I think, problem is in ESPAsyncWebServer core
Please Help me...

P.S.

void test() {
  HTTPClient myhttp;
  myhttp.begin("http://192.168.43.208/led=1");
  int myhttpCode = myhttp.GET();
  if(myhttpCode == HTTP_CODE_OK) {
      String myRequest = myhttp.getString();
      Serial.println(myRequest);
  }
  myhttp.end();
}

me-no-dev, 2017-09-13:
are you calling test inside async callback?

topersober, 2017-09-13:
Yes.  I tried inside:
1) server.on("/test", HTTP_GET, [](AsyncWebServerRequest * request) {
    test();  
  });

2) void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len) {
    test();
}

me-no-dev, 2017-09-13:
this would not work inside async callback. And in your callbacks you never respond to that request, which is another error. Since HTTPClient uses yield and delay underneath, that will break async and lead to reboot usually.

closed 2019-10-05 by stale[bot]