me-no-dev/ESPAsyncWebServer #231: Pages with similar names

terryjmyers, 2017-09-21:
Found a really minor issue.
If you have multiple pages such as:

	HTTPserver.on("/Config", HTTPConfig);
	HTTPserver.on("/Config/Network", HTTPNetwork);
	HTTPserver.on("/Config/LiveData", HTTPConfigLiveData);
	HTTPserver.on("/Config/Time", HTTPConfigTime);

Then only the /Config page will be returned even if the url is ....Config/Network.  I understand from the documentation that the pages are added in the order that they are instantiated, and this was the reason, but it was still an undesired behavior.  I simply changed the first line to:

	HTTPserver.on("/Config/General", HTTPConfigGeneral);

and renamed my function.  Just thought I'd mention the limitation in case you wanted to correct it.

me-no-dev, 2017-09-22:
just add ```HTTPserver.on("/Config", HTTPConfig);``` last :)

me-no-dev, 2017-09-22:
it goes in the order of  addition, so would match to the other ones first :)

closed 2019-10-05 by stale[bot]