me-no-dev/ESPAsyncWebServer #232: How to Serial.print POST paramater

gusantor, 2017-09-22:
Hello

after posting some parameters with a form, I'm trying to know the values of them

Iterating through parameters, as in the examples, I can get the values with 

> request->getParam(i);

But asking parameter by name, I get errors

    AsyncWebParameter* paramUN = request->getParam("USERNAME",true,false);
    //calling any of the following makes ESP to crash with Fatal exception 28(LoadProhibitedCause):
    //epc1=0x4021949d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000000c, depc=0x00000000
    String uN = paramUN->value();
    String uN = paramUN->value().c_str();

so, how should I proceed to know parameter value using it's name ?

thank you

me-no-dev, 2017-09-25:
have you checked this: https://github.com/me-no-dev/ESPAsyncWebServer#get-post-and-file-parameters

gusantor, 2017-09-25:
Hello, please sorry me

I was forgotting to wrap my code with an if hasParam check, since I added it, error it's gone

    if(request->hasParam("USERNAME") ){
      AsyncWebParameter* paramUN = request->getParam("USERNAME",true);
      Serial.println(paramUN->value());
    }

by the way, sure, from there I got valuable info

thanks

closed 2019-10-05 by stale[bot]